PX4/PX4-Autopilot #7032: Has anyone tested the offboard mavros control code?

vinhk, 2017-04-12:
I am testing it with Px4flow on my Pixhawk. I also added a few lines to see if it responds to movements commands. However, it goes off higher and higher when x or y is sent. Here is my code: 
/**
 * @file offb_node.cpp
 * @brief offboard example node, written with mavros version 0.14.2, px4 flight
 * stack and tested in Gazebo SITL
 */
```
#include <time.h>
#include <ros/ros.h>
#include <math.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>
#include <std_msgs/String.h> 
#include <stdio.h>
#include <math.h>
#include "geometry_msgs/Vector3Stamped.h"
#include "std_msgs/Float64.h"

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv)
{
    time_t start = time(0);
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;
    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(20.0);

    // wait for FCU connection
    while(ros::ok() && current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped pose;
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 3;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    geometry_msgs::PoseStamped cmd_att;

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request = ros::Time::now();

    while(ros::ok()){
        if( current_state.mode != "OFFBOARD" &&
            (ros::Time::now() - last_request > ros::Duration(5.0))){
            if( set_mode_client.call(offb_set_mode) &&
                offb_set_mode.response.success){
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {
            if( !current_state.armed &&
                (ros::Time::now() - last_request > ros::Duration(5.0))){
                if( arming_client.call(arm_cmd) &&
                    arm_cmd.response.success){
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

	
	
	double seconds_since_start = difftime( time(0), start);
	printf("Dur [ms]: %.1f \n", seconds_since_start);
	if(seconds_since_start == 25){
		ROS_INFO("25secs");
		pose.pose.position.x = 1;
		ros::spinOnce();
	}

	if(seconds_since_start == 35){
		ROS_INFO("35secs");
		pose.pose.position.y = -2;
		ros::spinOnce();
	}

	if(seconds_since_start == 45){
		ROS_INFO("45secs");
		offb_set_mode.request.custom_mode = "POSCTRL";
		ROS_INFO("Offboard enabled");
		ros::spinOnce();
	}
	local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
	//send a few setpoints before starting
   
    }

    
    return 0;
}
```

Stifael, 2017-04-13:
What you mean with 25 seconds? Did you try to test it first in sitl?

vinhk, 2017-04-13:
@Stifael once the the timing hit 25 seconds, I send the x position to see if the Pixhawk move and so on for y. I tested it in Gazebo and it worked well.

tomatac, 2017-04-14:
@vinhk I would try a time interval in the if statement and publish x y and z. Better, I would test if you reached position within a certain threshold, also adding a condition to stop if position has not been reached in a certain time.

dagar, 2017-04-16:
Is there a PX4 issue here? If not can you ask these types of questions on discuss (http://discuss.px4.io/) or gitter (https://gitter.im/PX4/Firmware)?

closed 2017-04-16 by dagar