PX4/PX4-Autopilot #7033: Bug in HIL_STATE_QUATERNION handling

mzahana, 2017-04-12:
Hi,
@TSC21 and I are developing HIL plugins in MAVROS. This is to allow any ROS-compatible simulator to be used in HIL setup.

In the decoding of HIL_STATE_QUATERNION msg, in PX4 firmware, specifically the ```vehicle_global_position```,
https://github.com/PX4/Firmware/blob/ecb2511a7bbad094bff23dd95036eda0656778f0/src/modules/simulator/simulator_mavlink.cpp#L402-L413
altitude is divided by ```1E2```, but ```lat/lon``` are not scaled down properly. In the msg definition,
https://github.com/PX4/Firmware/blob/912ed98a28022504ef3ec4bf1bce0fe041f83ac4/msg/vehicle_global_position.msg

```lat/lon/alt``` are assumed to be in deg/deg/meters. And, they are supposed to be received in deg*1E7/ and meters*1E2 according to 
https://pixhawk.ethz.ch/mavlink/#HIL_STATE_QUATERNION

So, I guess that 
https://github.com/PX4/Firmware/blob/ecb2511a7bbad094bff23dd95036eda0656778f0/src/modules/simulator/simulator_mavlink.cpp#L407-L408
should be divided by 1E7.

Thanks.

mhkabir, 2017-04-12:
> Also the vehicle_global_position msg does not have timestamp field,

> So, possibly another mistake here,
https://github.com/PX4/Firmware/blob/ecb2511a7bbad094bff23dd95036eda0656778f0/src/modules/simulator/simulator_mavlink.cpp#L404

Timestamp field is autogenerated. All messages have it.

mzahana, 2017-04-12:
I see. What about lat/lon scaling?

mhkabir, 2017-04-12:
Yes, that looks incorrect. Can you please send in a pull request?

mzahana, 2017-04-13:
A PR is created https://github.com/PX4/Firmware/pull/7039

mzahana, 2017-04-13:
#7039

closed 2017-04-24 by mzahana