PX4/PX4-Autopilot #7035: Snapdragon Flight params sync and save does not work

bharath374, 2017-04-13:
The following issues were observed on a very recent test of the latest PX4 code with the Snapdragon Flight platform:
- Parameters that get updated from QGroundControl do not persist across reboots. And no parameter file gets created in the /usr/share/data/adsp folder on target.
- Parameters do not sync between the apps processor and aDSP. When a parameter is updated from QGroundControl, it does not seem to reach the flight stack on the aDSP.

LorenzMeier, 2017-04-13:
Since this is passing through muORB it would be great if someone at Qualcomm could look into this. I also think that @darioxz can help with this in the long-run. It would be great if you could share your Jenkins setup with him, as he is building a whole hardware-test farm with Jenkins right now.

jywilson, 2017-04-15:
@LorenzMeier I am looking at this now.  It's a top priority problem for us.  

I will update this post as I make progress.

jywilson, 2017-04-19:
@LorenzMeier 
This problem is fixed in PR: https://github.com/PX4/Firmware/pull/7084

LorenzMeier, 2017-04-30:
Fixed.

closed 2017-04-30 by LorenzMeier