PX4/PX4-Autopilot #7040: Compile error with -Weffc++

bkueng, 2017-04-13:
On current master I get the following compile error:
```
../src/lib/matrix/matrix/Vector.hpp:20:7: error: base class 'class matrix::Matrix<float, 3u, 1u>' has a non-virtual destructor [-Werror=effc++]
 class Vector : public Matrix<Type, M, 1>
```

This is a false alarm and got introduced with https://github.com/PX4/Matrix/commit/66e1b406b8a68422135c365d42b987a64c9ddf8d

In general I agree with Scott Meyer's recommondations, but I'm starting to doubt the usefulness of the GCC flag. Also since we have clang-tidy that does similar things. @dagar what do you think?

dagar, 2017-04-13:
Yes, I agree. Are you hitting this with a different version of GCC locally or a different set of flags?

bkueng, 2017-04-13:
I currently use GCC 4.8.4, without any other local changes.

ndepal, 2017-04-13:
Just ran into this too on OSX, GCC 4.8.4. Is there a quick fix? Where do I disable that flag?

dagar, 2017-04-13:
Alright let's get rid of it. I need to see why the build system didn't catch this.

https://github.com/PX4/Firmware/pull/7044

PR #7044 (2017-04-13): cmake remove -Weffc++
 - closes #7040

@bkueng @ndepal FYI
commit 827725fb (2017-04-13): cmake remove -Weffc++ (#7044)

- closes #7040
closed 2017-04-13 by dagar