PX4/PX4-Autopilot #7050: HILstar (Xplane) model throwing sensor errors with recent PX4 Master

Antiheavy, 2017-04-14:
I just started HIL testing recent PX4 Master firmware (with small fixed wing tweaks).  The Xplane HIL mostly works and I can get it to fly, however, I get a lot of sensor errors thrown in QGC.  I change the plane model and calibrated the sensors and then changed back to HILstar(Xplane) - still get these sensor errors.  Pics below.

Any ideas?

[image]

dagar, 2017-04-16:
The first screenshot is QGC checking params that aren't set. Open another issue over there.

The timeout is from the PX4 sensors module and already relaxed for HIL> https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L86

@bkueng can we further increase the accel and gyro timeout for HIL?

DonLakeFlyer, 2017-04-16:
I won't go into a rant about how this optional param stuff has been making be crazy for years.

bkueng, 2017-04-18:
> @bkueng can we further increase the accel and gyro timeout for HIL?

Sure. Increase to however large is needed. But if there's really so much jitter, flight behavior will be affected negatively.

dagar, 2017-04-18:
Does it happen continuously, or just after the initial connection?

Antiheavy, 2017-04-18:
Sensor TOUT warnings happen:

- sometimes with initial connection, but not always.

- almost always when attempting to Arm/takeoff.
[image]

- occasionally throughout the HIL flight, sometimes when uploading missions.
[image]

- often when landing in HIL:
[image]

- almost always when disconnecting from the HIL sim (expected i suppose?).

PR #7129 (2017-04-25): sensors HIL increase gyro and accel timeout
 - fixes #7050

Probably worth taking a look at the QGC X-Plane interface to see if this can be improved.
commit f04f1d6b (2017-04-25): sensors HIL increase gyro and accel timeout

 - fixes #7050
closed 2017-04-25 by LorenzMeier