PX4/PX4-Autopilot #7055: commander rewrite

dagar, 2017-04-15:
Commander has become unmanageable. After the next stable release goes out let's decide on a path forward for a rewrite or major restructuring.

@LorenzMeier @bkueng 

I believe https://github.com/PX4/Firmware/pull/6750 is the only major commander work in progress.

Relevant - https://github.com/PX4/Firmware/pull/6955

priseborough, 2017-04-19:
In addition to the above notes and ideas:

- A specification for the intended state machine logic is created and reviewed before we start coding.
- Documentation is significantly improved on current practice. All class variables should be documented. The definitions of variables used for navigation states and transition events  are particularly important.
- Transition events are represented at the top level using a single struct. This can be comprised of smaller structs to separate different types of events - RC, failsafe, etc. The use of a unions with flags and integer bitmask will be efficient with storage.

mcharleb, 2017-04-19:
+1 for "A specification for the intended state machine logic is created and reviewed before we start coding"

ChristophTobler, 2017-05-05:
> * abstract the gps position to just has_position. It should not matter what the position source is (gps, vision, ...), as much as that is possible.

this is very important for the future, I think.

LorenzMeier, 2017-07-21:
Will bring this one in tomorrow: https://github.com/PX4/Firmware/pull/7517

LorenzMeier, 2017-08-07:
State machine testing: https://github.com/PX4/Firmware/tree/master/src/modules/commander/commander_tests

dagar, 2019-02-13:
Update for anyone interested in this.

There's likely never going to be a full "commander rewrite", but gradually we are getting the complexity under control. The biggest change I want to push for this year is finding some kind of framework or library for handling the state machine (https://github.com/PX4/Firmware/issues/10584).

davids5, 2019-02-13:
@dagar how about ESTELLE https://books.google.com/books?id=-OJ6CwAAQBAJ&pg=PA313&lpg=PA313&dq=estel+state+transition&source=bl&ots=-m7VzTVeGG&sig=ACfU3U3UsKzhqPvd6-J213YKpctpNGNpAA&hl=en&sa=X&ved=2ahUKEwjV--nmgrngAhUUJDQIHThyDUAQ6AEwAHoECAIQAQ#v=onepage&q=estel%20state%20transition&f=false

Or is that too old-school?

davids5, 2019-02-13:
Here is a more complete reference http://www-lor.int-evry.fr/idemcop/uk/est-lang/download/estelle-tutorial-slides.pdf