PX4/PX4-Autopilot #7059: no_sim=1 make posix_sitl_default gazebo not work

chuyueqiang, 2017-04-16:
hi,everybody.i want to do the gazebo simulation inferfacing to ROS,what i do is follow the page https://dev.px4.io/simulation-ros-interface.html ,but when i do the command
no_sim=1 make posix_sitl_default gazebo
the console output
[image]

the OS is ubuntu x64 14.04
ROS rev is indigo
gazebo rev is 7
i have already installed libprotobuf-dev,but it still not work
so,please help me to slove the problem,thank you

mhkabir, 2017-04-16:
Try `sudo apt-get install protobuf-compiler` in addition.

mhkabir, 2017-04-16:
Did you even follow this page? https://dev.px4.io/starting-installing-linux.html

chuyueqiang, 2017-04-16:
@mhkabir yes,all the page command i have done.and the protobuf-compiler i have already installed.but i try again just now,it also doesn't work.

dagar, 2017-04-16:
Did you install the dependencies before you tried to build the first time? I'm wondering if the gazebo build is stuck without having found the protobuf compiler. Try rebooting to make sure your path is updated. Then in the PX4 source do `make distclean`, then try to build again.

mhkabir, 2017-04-25:
@chuyueqiang Updates? Otherwise we will close this.

LorenzMeier, 2017-08-20:
Should be fine.

closed 2017-08-20 by LorenzMeier