PX4/PX4-Autopilot #7063: tailsitter after transiton to FW,can not control, has been flying straight line

MikeWellC, 2017-04-17:
make posix_sitl_ekf2 gazebo_tailsitter successful ,select mission mode and arm,tailsitter takeoff and transition,then can not control,fly straight line.make posix_sitl_default gazebo_tailsitter also same.what's wrong,please help me!

[ 84%] Built target lib__runway_takeoff
[ 85%] Built target lib__version
[ 85%] Built target lib__terrain_estimation
[ 85%] Built target lib__tailsitter_recovery
[ 86%] Built target platforms__posix__px4_layer
[ 87%] Built target modules__bottle_drop
[ 88%] Built target examples__rover_steering_control
[ 91%] Built target platforms__posix__work_queue
[ 92%] Built target examples__segway
[ 92%] Built target examples__px4_simple_app
[ 93%] Built target examples__px4_daemon_app
[ 93%] Built target examples__px4_mavlink_debug
[ 93%] Built target examples__fixedwing_control
[ 94%] Built target modules__events
[ 95%] Built target modules__fw_att_control
[ 95%] Built target modules__fw_pos_control_l1
[ 98%] Built target modules__systemlib
[100%] Built target px4
[100%] Built target px4_tailsitter
args: /home/mike/src/Firmware/build_posix_sitl_ekf2/src/firmware/posix/px4 posix-configs/SITL/init/ekf2 none gazebo tailsitter /home/mike/src/Firmware /home/mike/src/Firmware/build_posix_sitl_ekf2
SITL ARGS
sitl_bin: /home/mike/src/Firmware/build_posix_sitl_ekf2/src/firmware/posix/px4
rcS_dir: posix-configs/SITL/init/ekf2
debugger: none
program: gazebo
model: tailsitter
src_path: /home/mike/src/Firmware
build_path: /home/mike/src/Firmware/build_posix_sitl_ekf2
GAZEBO_PLUGIN_PATH /home/mike/src/Firmware/build_posix_sitl_ekf2/build_gazebo:
GAZEBO_MODEL_PATH :/home/mike/src/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/home/mike/src/Firmware/Tools/sitl_gazebo/Build/msgs/:/home/mike/src/Firmware/build_posix_sitl_ekf2/build_gazebo
SITL COMMAND: /home/mike/src/Firmware/build_posix_sitl_ekf2/src/firmware/posix/px4 /home/mike/src/Firmware /home/mike/src/Firmware/posix-configs/SITL/init/ekf2/tailsitter
data path: /home/mike/src/Firmware
commands file: /home/mike/src/Firmware/posix-configs/SITL/init/ekf2/tailsitter
18446744073709551501 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

Gazebo multi-robot simulator, version 7.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 7.5.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.125.54
[Msg] Publicized address: 192.168.125.54
INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11797680 bytes
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [rotor_0_joint] found for channel[0] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [rotor_1_joint] found for channel[1] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [rotor_2_joint] found for channel[2] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [rotor_3_joint] found for channel[3] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [left_elevon_joint] found for channel[5] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [right_elevon_joint] found for channel[6] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:158] joint [elevator_joint] found for channel[7] joint control active for this channel.
[Err] [Scene.cc:2941] Light [sun] not found. Use topic ~/factory/light to spawn a new light.
INFO  [simulator] Got initial simuation data, running sim..
INFO  [pwm_out_sim] MODE_8PWM
WARN  [vtol_att_control] failed setting min values
.
.
INFO  [mavlink] mode: Normal, data rate: 2000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 2000000 B/s on udp port 14557 remote port 14540
INFO  [sdlog2] [blackbox] rootfs/fs/microsd/log/2017-04-17
WARN  [sdlog2] sdlog2: failed setting sched params
INFO  [sdlog2] [blackbox] recording: 01_37_16.px4log
INFO  [tone_alarm] startup
Command 'sdlog2' failed, returned 1
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
pxh> INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 21, OBS buf: 16)
INFO  [mavlink] partner IP: 127.0.0.1
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/tailsitter/motor_speed/0, deleting message. This warning is printed only once.
INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
INFO  [lib__ecl] EKF commencing GPS fusion
INFO  [commander] home: 47.3977419, 8.5455938, 487.84
INFO  [tone_alarm] home_set
WARN  [commander] mike test 1 navmode 4
INFO  [tone_alarm] neutral

dagar, 2017-04-18:
I can look at this tomorrow. I'm guessing it's easy to reproduce, but if you have any logs to share they might be helpful.

MikeWellC, 2017-04-18:
thank you,sitl where is the logs location? @dagar

dagar, 2017-04-18:
They're buried in the build directory (rootfs/fs/microsd/log/2017-04-17). It's probably easier to just do `find . -name *.px4log` 
Uploading to http://logs.uaventure.com/ is a good way to share.

MikeWellC, 2017-04-18:
this is a log i just done ,tailsitter just respond to "Emergency stop".
http://logs.uaventure.com/view/xfVo2M6TQjySFP6FFviu3R @dagar

bresch, 2017-04-19:
I tried and have the same issue. My mission works with the standard VTOL but fails with the tailsitter. 
It's acknowledging the commander commands but nothing works when in FW mode. Back transition isn't possible. As MikeWellC said, it just responds to Emergency stop.

dagar, 2017-05-02:
It's stuck in transition waiting to achieve airspeed and there's currently no timeout implemented. I've added one, but need to do some testing.
The tailsitter gazebo simulation doesn't work right because it's not registering airspeed when moving forward in the FW frame. We'll probably need to hack it in sitl_gazebo.

sanderux, 2017-06-09:
@dagar i implemented the timeout here: https://github.com/PX4/Firmware/pull/5185
but i dont have a tailsitter so was waiting for a test

dagar, 2017-06-09:
Thanks @sanderux. We might as well get it working reasonably well in SITL and get it in. The current code is broken for several real world tailsitters I've encountered. There's also no pitch actuation during P2 right now which we should change.
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/tailsitter.cpp#L501

dagar, 2017-06-27:
It's getting stuck in transition waiting for the airspeed threshold.
 1. need a transition timeout for tailsitter
 2. figure out how to get tailsitter airspeed for gazebo. I'm guessing this is why it working before. https://github.com/PX4/sitl_gazebo/pull/69

LorenzMeier, 2017-06-28:
We should just revert that change.

LorenzMeier, 2017-08-20:
This is fixed.

closed 2017-08-20 by LorenzMeier