PX4/PX4-Autopilot #7066: tailsitter drop down

MikeWellC, 2017-04-17:
http://logs.px4.io/plot_app?log=3afcc56c-5b9d-415c-ab46-43c8fe700b6e
my tailsitter fell from the air,I adjusted this parameter FW_P_TC to 0.4 and then drop down.I have some questions.
1.What is the relationship between the magnetic field and the thrust?
I understand that thrust should not affect the magnetic field.is it?
2.What is the reason my tailsitter is falling off?
3.How should I adjust the parameters?

dagar, 2017-06-27:
Hi @MikeWellC, sorry for the delayed response.

Have you had successful tailsitter flights?

Is this extreme vibration from your motor?

[image]

MikeWellC, 2017-07-01:
thank  @you，I found the reason that my pixhawk was loose and vibrated inside the tailsitter。

MikeWellC, 2017-07-01:
Hi @dagar, thank you，I found the reason that my pixhawk was loose and vibrated inside the tailsitter。

dagar, 2017-07-01:
Glad to hear you found the problem.

closed 2017-07-01 by dagar