PX4/PX4-Autopilot #7078: OSD on TELEM1?

mzahana, 2017-04-19:
Hi,

I know there is options in QGC to setup TELEM2 port to send OSD data. However, some people use it as a companion link at 921600, and keep TELEM1 for telemetry. Is there a way to have options parameter for TELEM1 just like TELEM2. That would give more flexibility.

Thanks.

dagar, 2017-04-20:
Does an OSD work with the default telemetry configuration on 1? That's 57600 baud and the "normal" set of messages.

mzahana, 2017-04-20:
What I know is that TELEM1 does not send OSD.

KJNKJNKJN, 2017-04-24:
hello ,friends .i have a question .i want to change the  baud of TELEM2 from 57600 to 115200 .but 115200 doesn't exist in the qgc .what should i do ?

mzahana, 2017-04-24:
@KJNKJNKJN I can see it's already handled here,
https://github.com/PX4/Firmware/blob/da8724accb2fe83ebee82c9db9fc9f8f1680db6b/ROMFS/px4fmu_common/init.d/rcS#L702
 but the option is not included in QGC yet. You might want to try to set it manually through NSH
``` nsh> param set SYS_COMPANION 3115200```. Then, reboot, and check.

sanderux, 2017-05-01:
@dagar OSD mode just streams less: https://github.com/PX4/Firmware/blob/45af77a5431a35444afd4d80e32aabc8feb36c66/src/modules/mavlink/mavlink_main.cpp#L2048

so it should work just as well on telem1

dagar, 2017-05-02:
WIP PR to add options for the default interface. https://github.com/PX4/Firmware/pull/5184

closed 2017-05-02 by dagar