PX4/PX4-Autopilot #7083: LPE/MC_POS_CTL: Altitude hold fails after LIDAR fault without recovery

ecmnet, 2017-04-19:
Being in ALTHOLD and trying to keep the quad at the same position manually (latest master, PixRacer, LPE)  at about 1m for several seconds, the quad shoots up in the air, without any further response to the sticks. To avoid further climb, I had to switch off by the kill switch causing the vehicle to crash, which disconnected power. Although log should have been written to log004.ulg, it has not been stored on SD - so unfortunately no log is available, but messages were written to disc:

2000_01_01_00_34_17: ARMED by RC
2000_01_01_00_34_17: [logger] file: /fs/microsd/log/sess018/log004.ulg
2000_01_01_00_34_21: Takeoff detected
2000_01_01_00_34_21: [lpe] land timeout 
2000_01_01_00_34_24: [lpe] lidar init: mean 21 cm stddev 10 cm
2000_01_01_00_34_26: [lpe] flow init: quality 244 std 9
2000_01_01_00_34_32: [lpe] flow timeout 
2000_01_01_00_34_33: [lpe] flow init: quality 232 std 24
2000_01_01_00_34_52: [lpe] lidar fault,  beta 10.79
2000_01_01_00_34_53: MANUAL KILL SWITCH ENGAGED
2000_01_01_00_34_53: Failsafe enabled: no gps
2000_01_01_00_34_53: Failsafe mode enabled
2000_01_01_00_34_54: [lpe] flow timeout 
2000_01_01_00_34_54: Autopilot off, returned control to pilot

As noted earlier, I had experienced this several times now since at least 1.5.

ecmnet, 2017-04-20:
Update: Leaving the vehicle on ground for a while (land detector not activated) and then rising it up manually produces the following effect:

2017-04-20 10:12:01.261: [6] [lpe] lidar init: mean 20 cm stddev 11 cm
2017-04-20 10:12:01.362: [6] [lpe] lidar fault,  beta 328.16
2017-04-20 10:12:23.564: [6] [lpe] lidar timeout
2017-04-20 10:12:39.762: [6] [lpe] lidar init: mean 29 cm stddev 19 cm
2017-04-20 10:12:39.966: [6] [lpe] lidar fault,  beta 318.56
2017-04-20 10:12:50.761: [6] [lpe] lidar timeout

In this case after fault with beta at 318.56, lidar did not recover. I assume this was due to baro drift on ground. A similar situation, in my point of view, could be a possible reason for the issue described above.

jgoppert, 2017-04-28:
Are you using all of the default params for lidar noise etc?

ecmnet, 2017-04-29:
Yes, for LIDAR all settings were default, but I increased BARO to reduce it's effect.

ecmnet, 2017-04-30:
@jgoppert: Let me try an explanation: Leaving the vehicle powered for a while on ground, baro drifted away. As baro does not seem to be initialized, e.g. at takeoff, there is a huge bias between LIDAR and baro. When LIDAR failed in AltHold-mode, altitude is corrected by this bias. Maybe that this rapid movement caused a LIDAR covariance that is to high to re-initialize, when LIDAR got available again.

The reason, why LIDAR failed seems to be another issue. Meanwhile I switched back to PixHawk-V2 without having any LIDAR dropouts anymore.

ecmnet, 2017-04-30:
Directly after PX4 reboot:
[image]

After 1/2h leaving the vehicle on ground (in this case baro drift is negative):
[image]

closed 2018-02-21 by PX4BuildBot