PX4/PX4-Autopilot #7094: Changing PX4IO_SERVO_COUNT to 12 bricks Pixhawk?

imcnanie, 2017-04-21:
Hello, I changed PX4IO_SERVO_COUNT from 8 to 12 in px4io.h and it permanently bricked my px4 servo outputs. I tried this again on my second pixhawk, and it bricked that one too!

It still connects to QGC, but even after programming it with ardupilot and px4, there is still nothing on the servo outputs. Why is this? For context, I am working on using the AUX channels controlled by px4fmu as 4 extra outputs available with px4io.

mhkabir, 2017-04-21:
I think force-reflashing the IO board should fix it.

LorenzMeier, 2017-04-21:
To that, press the safety switch, then power on the Pixhawk, then let it go. It should auto-update

imcnanie, 2017-04-24:
@LorenzMeier That fixed both of them! Any idea why this happens?

LorenzMeier, 2017-04-24:
You uploaded bad code and bricked the application binary. That method avoids ever running your (broken) application binary and flashes the default instead.

closed 2017-04-24 by LorenzMeier