PX4/PX4-Autopilot #7103: AUX Broken on Master

tops4u, 2017-04-21:
I have set the Camera Trigger to use GPIO on Pin 3 some time ago, but never actually never used the camera trigger.

I installed Master today and then my AUX Ports stopped working all exhibiting 0us PWM. Then I set the Camera trigger back to PWM and removed all Trigger Pins just after this the AUX Outputs gave a PWM Signal. 

IMHO it should be possible to use a few pins as camera GPIO Pins and keep the rest as AUX PWM Outputs.

sanderux, 2017-04-22:
@mhkabir 
related to https://github.com/PX4/Firmware/pull/6973 ?

mhkabir, 2017-04-22:
No, this is by design. 

If you want to use the AUX pins as actuator outputs, and at the same time want to use triggering, you need to use trigger on pins 5 or 6 due to how the timers can be used on the hardware. 

I will document this in the user guide soon. Waiting to construct a full mapping setup, so that I do the docs for trigger + geotag in one go.

closed 2017-04-22 by mhkabir