PX4/PX4-Autopilot #7105: Latest 1.6.0 RC release not supported on Pixhawk 3 Pro board (Drotek)

HuguesD, 2017-04-22:
Hello,

Just got a recently produced Pixhawk 3 Pro board from Drotek. The following error happens when attempting to load the firmware (px4fmu-v4_default.px4)
"downloaded firmware board id does not match hardware board id: 11!=13"

I'm aware that board id 11 is a pixracer board. Board id of the Drotek PixhawkV3 is 13. what firmware is compatible with this board id 13 ?

[image]

mhkabir, 2017-04-22:
You need the `px4fmu-v4pro_default` target. `v4` is Pixracer.

HuguesD, 2017-04-22:
Hello Kabir,

 

Thx. 

where can I find this version of the firmware, It is not on Github with the other PX4 firmwares ?

mhkabir, 2017-04-22:
The Pixhawk 3 Pro is not fully supported by the user-facing side of things yet. It is a developer system and is actively being worked on and fixed.

You can build and flash the firmware from the repo. Instructions here : https://dev.px4.io/en/

closed 2017-04-22 by mhkabir