PX4/PX4-Autopilot #7108: Lidar lite v3 sensor data integration problem

kevinlyu1, 2017-04-23:
I am currently working with the px4 and lidar lite v3 and can't seem to figure out how to get the lidar lite v3 to be fused with the altitude_local. I can see the lidar lite v3 data on the ALTITUDE.bottom_clearance, but it isn't being used. Does anyone know how to fix this?

I went through the nsh and was able to see the data through this method as well.

dagar, 2017-04-26:
Did you look at the LPE parameter LPE_FUSION? 
https://dev.px4.io/en/advanced/parameter_reference.html#local-position-estimator

kevinlyu1, 2017-04-27:
Haven't yet, but I think I'll try it. Ended up getting it working on another px4 that was using LPE so I'll try it on the original px4 and see if that does it.

dagar, 2017-04-28:
Alright, I'll close this for now. If you mind it's still an issue somewhere please reopen with any debugging information (logs, console output, etc).

closed 2017-04-28 by dagar