PX4/PX4-Autopilot #7109: Test and document camera test trigger

LorenzMeier, 2017-04-23:
We should make sure that this button in QGC daily works:

[image]

I have verified that the command arrives here:
https://github.com/PX4/Firmware/blob/3fda79ef8b05493ded0037c6bc5305f2b2563f72/src/modules/mavlink/mavlink_receiver.cpp#L171-L184

Needs further tracing through the system to ensure its being properly published and handled in the trigger module.

LorenzMeier, 2017-04-23:
When this gets fixed it should work again: https://github.com/mavlink/qgroundcontrol/issues/5045

mhkabir, 2017-04-24:
Fixed.

mhkabir, 2017-04-24:
Reopening till I get the docs in.

LorenzMeier, 2017-04-27:
@mhkabir Where is this standing?

mhkabir, 2017-04-28:
This is done from my side : https://dev.px4.io/en/tutorials/camera_trigger.html

I still need a new geotagging implementation in QGC with the new logger to document that part.

LorenzMeier, 2017-05-01:
I'm closing this since the geotagging is handled in a different issue.

closed 2017-05-01 by LorenzMeier