PX4/PX4-Autopilot #7112: Include I2C option for Lidar Lite v3

mzahana, 2017-04-23:
Hi,

I noticed that the option to enable Lidar Lite is still PWM. However, it does not work, and I think it's broken. I've seen some cases where Lidar Lite v3 worked on I2C, but the ```rsC``` file had to be modified here,
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L743

Since Lidar Lite v3 is taking over v2, and it works on I2C, then I think it's better to modify the startup command to use I2C, instead of PWM. Please advise. Thanks,

mhkabir, 2017-04-23:
Can you please add an option to the parameter [`SENS_EN_LL40LS`](https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensor_params.c#L3233-L3240) ( = 2 ), such that the startup script starts the I2C driver when set to 2. Leave in the PWM startup option for now.

Make sure you add the metadata to the param file properly, such that QGC can display the options. You can follow how it's done for the [SF0X](https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensor_params.c#L3242-L3256). 

Waiting for the PR :)

mzahana, 2017-04-23:
Done. #7113

closed 2017-04-24 by mzahana