PX4/PX4-Autopilot #7116: More than 8 rotors using multiple Pixhawks/Aux channels

imcnanie, 2017-04-24:
Hello, I'm planning to build a copter with 12 rotors. **Is there a simple way to do this with an aux mixer?** 

Another thought is to do it using the snapdragon_pwm_forwarder SYS_AUTOSTART, but that seems like that would require heavy modification to mixer.cpp.

dagar, 2017-04-26:
I'm not sure how to do this out of the box, but I can think of a few ways to hack it.

What if you generated the mixer for 12, but then manually split it into 2? Place them on control groups 0 and 1. https://dev.px4.io/en/concept/mixing.html

imcnanie, 2017-04-26:
Interesting. I don't completely understand how the control groups work yet. Mind explaining a bit more?

I have two other ideas, I'd love thoughts on them. Yesterday I wrote two mixers:

dodeca_main_outputs_cox = [
    [  90, CW],
    [ -90, CCW],
    [ -30, CW],
    [ 150, CCW],
    [  30, CCW],
    [-150, CW],
    [  90, CCW],
    [ -90, CW],

]

dodeca_aux_outputs_cox = [
    [ -30, CCW],
    [ 150, CW],
    [  30, CW],
    [-150, CCW],
]

set MIXER dodeca_main_outputs_cox
set MIXER_AUX dodeca_aux_outputs_cox

I connected the MAIN outputs to dodeca_main_outputs_cox and the AUX outputs to the dodeca_aux_outputs_cox mixer. This flew but lost control pretty quickly (I also adjusted PWM limits/rates etc).

**First idea**: Do the same thing but mix the bottom half of the copter as one hexacopter using px4fmu (AUX) and the top as another hexacopter using px4io (MAIN). Like this:

dodeca_main_outputs_cox = [
    [  90, CW],
    [ -90, CCW],
    [ -30, CW],
    [ 150, CCW],
    [  30, CCW],
    [-150, CW],
]

dodeca_aux_outputs_cox = [
    [  90, CCW],
    [ -90, CW],
    [ -30, CCW],
    [ 150, CW],
    [  30, CW],
    [-150, CCW],
]

I think this might have problems. Two mixers working without each other sounds like bad news. Maybe not though!

**Second Idea**: Use one AUX mixer, set the MAIN mixer to pass. Then forward the extra outputs over uORB from the px4fmu driver to the px4io driver.

dodeca_aux_outputs_cox = [
    [  90, CW],
    [ -90, CCW],
    [ -30, CW],
    [ 150, CCW],
    [  30, CCW],
    [-150, CW],
    [  90, CCW],
    [ -90, CW],
    [ -30, CCW],
    [ 150, CW],
    [  30, CW],
    [-150, CCW],
]

Let me know. I'm going to try the first approach right now.

imcnanie, 2017-04-26:
Although I believe px4io doesn't support orb.

dagar, 2017-04-26:
I would have thought splitting them would be equivalent, at least until you run into motor saturation? I probably don't understand the multicopter mixer well enough.

@Stifael @MaEtUgR any advice?

Here's the generated mixer tables for the entire dodeca coaxial, and then split into 2 groups. Wouldn't this result in the same motor output? 

```C
const MultirotorMixer::Rotor _config_dodeca_cox[] = {
        { -1.000000,  0.000000, -1.000000,  1.000000 },
        {  1.000000,  0.000000,  1.000000,  1.000000 },
        {  0.500000,  0.866025, -1.000000,  1.000000 },
        { -0.500000, -0.866025,  1.000000,  1.000000 },
        { -0.500000,  0.866025,  1.000000,  1.000000 },
        {  0.500000, -0.866025, -1.000000,  1.000000 },
        { -1.000000,  0.000000,  1.000000,  1.000000 },
        {  1.000000,  0.000000, -1.000000,  1.000000 },
        {  0.500000,  0.866025,  1.000000,  1.000000 },
        { -0.500000, -0.866025, -1.000000,  1.000000 },
        { -0.500000,  0.866025, -1.000000,  1.000000 },
        {  0.500000, -0.866025,  1.000000,  1.000000 },
};

const MultirotorMixer::Rotor _config_dodeca_cox_main[] = {
        { -1.000000,  0.000000, -1.000000,  1.000000 },
        {  1.000000,  0.000000,  1.000000,  1.000000 },
        {  0.500000,  0.866025, -1.000000,  1.000000 },
        { -0.500000, -0.866025,  1.000000,  1.000000 },
        { -0.500000,  0.866025,  1.000000,  1.000000 },
        {  0.500000, -0.866025, -1.000000,  1.000000 },
};

const MultirotorMixer::Rotor _config_dodeca_cox_aux[] = {
        { -1.000000,  0.000000,  1.000000,  1.000000 },
        {  1.000000,  0.000000, -1.000000,  1.000000 },
        {  0.500000,  0.866025,  1.000000,  1.000000 },
        { -0.500000, -0.866025, -1.000000,  1.000000 },
        { -0.500000,  0.866025, -1.000000,  1.000000 },
        {  0.500000, -0.866025,  1.000000,  1.000000 },
};
```

imcnanie, 2017-04-27:
Well that works! I am still interested in if there will be motor saturation issues. Also is there a performance hit by running the mixer on the fmu? Anything else I should be concerned with? @dagar

dagar, 2017-04-28:
It should be fine performance wise. Plenty of people use FMU only for multicopters with the pixracer. 

I'm not sure about motor saturation. If you're concerned you should review how multirotor motor limits are implemented and used in the multicopter attitude controller. Then pay close attention in the logs. With the current implementation I think you're only going to get the limit from either the IO or the FMU, but not both. 

https://github.com/PX4/Firmware/blob/c5e0bf1c2f4f2cfbc3aa95bfbb14c7b45778b1e0/src/modules/mc_att_control/mc_att_control_main.cpp#L164

If there's more to discuss let's continue at http://discuss.px4.io/. Please share video if you can!

closed 2017-04-28 by dagar