PX4/PX4-Autopilot #7120: make px4fmu-v2_test failing on master?

pythonmobile, 2017-04-24:
I am trying to switch on sf1xx support (to use SF11/C sensor), but even without switching it on, when I do :

 2179  git clone https://github.com/PX4/Firmware.git
 2180  cd Firmware/
 2181  git submodule update --init --recursive
 2182  make px4fmu-v2_default
 2183  make px4fmu-v2_test

I get the following error (can someone help me fix this? I tried to comment out a few sensors, did not help):
(Also I want to enable sf1xx in cmake/configs/nuttx_px4fmu-v2_default.cmake - and then be able to get rid of the following error)

[ 98%] Building CXX object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/circuit_breaker.cpp.obj
[ 98%] Built target romfs
[ 98%] Building CXX object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/battery.cpp.obj
[ 98%] Building CXX object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/hysteresis/hysteresis.cpp.obj
[ 98%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/err.c.obj
[ 98%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/printload.c.obj
[ 98%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/param/param.c.obj
[100%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/flashparams/flashparams.c.obj
[100%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/flashparams/flashfs.c.obj
[100%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/up_cxxinitialize.c.obj
[100%] Building C object src/modules/systemlib/CMakeFiles/modules__systemlib.dir/hx_stream.c.obj
[100%] Linking CXX static library libmodules__systemlib.a
[100%] Built target modules__systemlib
[100%] Linking CXX static library libmodules__uavcan.a
[100%] Built target modules__uavcan
Scanning dependencies of target firmware_nuttx
[100%] Building C object src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/builtin_commands.c.obj
[100%] Linking CXX executable firmware_nuttx
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: firmware_nuttx section `.text' will not fit in region `flash'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: address 0x81058c8 of firmware_nuttx section `__param' is not within region `flash'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: address 0x81058c8 of firmware_nuttx section `__param' is not within region `flash'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: address 0x81058c8 of firmware_nuttx section `__param' is not within region `flash'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `flash' overflowed by 28240 bytes
collect2: error: ld returned 1 exit status
src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/build.make:180: recipe for target 'src/firmware/nuttx/firmware_nuttx' failed
make[3]: *** [src/firmware/nuttx/firmware_nuttx] Error 1
CMakeFiles/Makefile2:7407: recipe for target 'src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all' failed
make[2]: *** [src/firmware/nuttx/CMakeFiles/firmware_nuttx.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:161: recipe for target 'px4fmu-v2_test' failed
make: *** [px4fmu-v2_test] Error 2

====================================================

How does one check which sensor is taking how much space on the flash?

LorenzMeier, 2017-04-24:
You need to disable a different module to make space in flash for the additional code built. Besides, you should only be using `make px4fmu-v2_default`, not the other make target.

closed 2017-04-24 by LorenzMeier

pythonmobile, 2017-04-25:
@LorenzMeier Thanks. How do I test whether all my modules fit in size? Also, even when I do just _default target, the SF11/C Lidar does not alter the Altitute readings in any way, when connected using I2C? Any thoughts on how to debug that?