PX4/PX4-Autopilot #7123: Motors spin when unplugging USB

ChristophTobler, 2017-04-25:
I cannot reproduce this consistently but here is what can happen:
I plug in the battery and USB to a Pixracer (latest master). If I then unplug the USB cable, it can happen that the rotors spin for a second which can be quite dangerous.

dagar, 2017-04-26:
Is the vehicle armed? Is the USB circuit breaker set?

ChristophTobler, 2017-04-26:
vehicle is not armed. What's the USB circuit breaker?

dagar, 2017-04-26:
To prevent arming with USB attached.

dagar, 2017-04-26:
Is the board resetting when USB is pulled?
@davids5 can anything be done here?

ChristophTobler, 2017-04-26:
Is that not a default? I haven't changed anything regarding USB circuit breaker so it should be default. Yes, the board is resetting

mhkabir, 2017-04-26:
OK this is interesting.

1. The board reset on USB disconnect is a feature.
2. If this started happening only recently, it might have something to do with the early timer init, @davids5. ESCs might be responding to transients in the signal at startup.

@ChristophTobler What ESCs are you using, and have you started seeing this behaviour only recently?

dagar, 2017-04-26:
Come to think of it, I think I've seen this on VTOLs where the FW is typically on the FMU. I've seen the pusher throttle engage when abruptly resetting power.

mhkabir, 2017-04-26:
@dagar Have you seen this only recently, or is it older? Because if its new, it's highly likely that it's from the recent FMU changes from me and David. We need to get a logic analyser on the pins during reboot to see what happens to the signal.

dagar, 2017-04-26:
Definitely not new.

davids5, 2017-04-27:
> Is the board resetting when USB is pulled?
 @davids5 can anything be done here?

I am not aware of board reset on USB disconnect as an added feature? 

I will get this on the bench an have a look.

Suspect is the Pixracer only feature in rcS to start the FMU stop the FMU for sensor reset. I would rather move that to the board init.

bkueng, 2017-04-27:
Now that I see it, I also experienced this, but several weeks/months ago, on a Pixracer.

@davids5 Reset is done in commander: https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2030

I currently have some changes to the fmu, including that it will not start the fmu when doing a sensor reset. I'll push that soon.

MaEtUgR, 2017-04-27:
I had the exact same problem when setting up a pixracer with master two weeks ago, this was pretty scaring, luckily I didn't have the propellers attached at this very moment.

dagar, 2017-04-27:
Why does it cause disarmed ESCs connected to the FMU to spin up, but not the ones connected to the IO?

ndepal, 2017-04-27:
I've also seen this happen on a Pixfalcon, on 21bc97b5bba983c52589b30c6d1e21a8c91575ff

LorenzMeier, 2017-04-27:
Definitely gating the release. @bkueng do you have an update on the FMU driver?

ndepal, 2017-04-27:
Would those FMU changes have any effect on systems with an IO?

bkueng, 2017-04-27:
Steps to reproduce:
- run QGC with auto-connect
- Start RC receiver (not sure if that's really needed)
- Plug vehicle to battery
- Plug in USB -> connect QGC
- Plug out USB -> system reboots & motors spin, but the spinning does not happen always (maybe around 20% or less of the cases)

Even with the 'do not start the fmu for sensor reset' change it still happens.

@ndepal No they should not.

ndepal, 2017-04-27:
So I guess its not an FMU specific problem.

I do not need to turn on the RC receiver to reproduce this.

davids5, 2017-04-28:
@bkueng No Joy in getting failure. It may be because I do not have ESC just the Logic analyzer. I do not have any PWM ESC and I am wondering if there is a back feed issue, that I can not create without ESC.  
What 

Typical capture: 

[image]

Left side zoomed:
[image]

Right side zoomed:
[image]

bkueng, 2017-04-28:
Update:
- removing `fmu sensor_reset` does not help
- it happens right on reboot, together with the output:
```
nsh> WARN  [commander] USB disconnected, rebooting.
WARN  [mavlink] Failed to open MAVLink log: /fs/microsd/msgs_2000_01_01_00_13_37.txt
�sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
```
  So before the start of any module.
- I increased the sleep time in https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2037 to see if it's the USB disconnect or reboot, and it's the reboot that triggers the spinning.

dagar, 2017-04-28:
@bkueng what's your setup? I've had this happen randomly in the past, but tried (and failed) to reproduce it last night.

bkueng, 2017-04-28:
I use a quad with Pixracer on battery with RC & gps attached, with 12A SimonK AutoShot ESC's. No special param config otherwise.

I did more testing and it turns out the USB is not needed: a `reboot` in NSH on battery without USB connected triggers the spinning too.

davids5, 2017-04-28:
I pushed this branch https://github.com/PX4/Firmware/pull/7151, It cuts the servos before the reset.  @bkueng  tested it and it did not fix it. 
So something is happen on system init. But I can not capture it.  I will get 4 ESC ASAP.

@bkueng what is you auto start?

bkueng, 2017-04-28:
> @bkueng what is you auto start?

4001

sanderux, 2017-05-01:
@ChristophTobler do you have a (5v) red wire attached to your esc's?
@bkueng those autoshot esc's seem to have a bec too, does it also happen when you cut the red wires?

davids5, 2017-05-02:
I received the ESC's and have captured the cause.

Setup is 2 Lumenier 12 amp ESC w/ SimonK AutoShot (2-4s N-FET) FMU_CH1 and FMU_CH2
Motors SunnySky X1806S KV2300

With Both Red wires connected on FMU_CH1 and FMU_CH2

[image]

[image]

With One  Red BEC wires connected FMU_CH1
[image]

With NO  Red BEC wires connected
[image]

The only thing that stops the spin (tested 20 times) is (go figure) adding 2 channels of a 16 channel Saleae (Non Pro) - I suspect it loads the lines and discharges them more rapidly.

To validate my supposition, I added 2 15K pull downs resistors on each of the FMU_CH1 and FMU_CH2 to ground.  I have repeated the reboot >20 times and no spin.

@bkueng would you be able to place some 15K  through hole resistors on FMU_CH1 and FMU_CH2 to ground to cross check?

LorenzMeier, 2017-05-02:
@davids5 It makes no sense modifying hardware for @bkueng. If you cut the red wires (power), does it still do that?

davids5, 2017-05-02:
> With NO Red BEC wires connected

@LorenzMeier  see Last pic it is with no red wires.

In the AM I can try to change the io cinfiguration

davids5, 2017-05-02:
@LorenzMeier - I got! 

Master 2ef5ebb6dbd9f1f9617ecb60e5b976fe23209bbd decay time is 540uS
[image]

PR that fixes it is [here](https://github.com/PX4/Firmware/pull/7151) decay time is 13.8 uS 
[image]

LorenzMeier, 2017-05-03:
Fixed on master.

closed 2017-05-03 by LorenzMeier