PX4/PX4-Autopilot #7125: Format pre-commit hook does not check what you actually commit

MaEtUgR, 2017-04-25:
You can have style errors that are not staged and you cannot commit because of the hook.
You can have style fixes unstaged and commit without them.

It's just a glitch that prevents you from being sure that what you commit is actually checked.

dagar, 2017-04-26:
Our format checking is pretty basic and yet still a convoluted set of scripts. We'd be better off investing the effort switching to clang-format where these things are already implemented properly.

For the time being you can use `git commit --no-verify` to skip the hook.

MaEtUgR, 2017-05-02:
The problem is not that I want to commit without check, rather the opposite I want to be sure if the check runs over my commit it has no style problems.

A beginner might not understand this problem in the first place and even an experienced user might forget to stage (`git add`) the changes proposed by `make format`.

The summarized result is: the style was fixed by `make format`, the fixes are not staged, you commit the old version without fixes and the hook does not realize it. Maybe you push, then CI fails, then you see that there were unstaged changes, you have to recommit and all of that could have been prevented.

I talked to @Stifael and at least I'm not the only one who had this experience before.

My guess was that it is a pretty easy fix for someone familiar with the checking script to check only the staged changes using some git magic. Maybe I will look into that myself if I get upset enough but I think it is woth this issue to not forget.

dagar, 2017-05-02:
The pre-commit hook just runs the checker against files that have changed (on disk). We'd need to pipe the staged version instead. https://github.com/PX4/Firmware/blob/master/Tools/pre-commit#L59

Although I'm still more interested in moving to clang-format if we're going to do anything. It does a better job, has more options, and integrates with virtually every editor/IDE we use.

closed 2018-02-21 by PX4BuildBot