PX4/PX4-Autopilot #7127: Master 0d22e97c09e493559b8c9645bea21e4cd7c53077 out of SRAM

davids5, 2017-04-25:
px4fmu-v3_defualt Running on FMUv2 2Mib capable HW. Note: *** below

```
nsh> reboot
ÿ[boot] Fault Log info File No 3 Length 3257 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
[hardfault_log] Fault Log is Armed
INFO  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
MS5611_SPI on SPI bus 4 at 2 (20000 KHz)
WARN  [ms5611] no device on bus 4
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [ver] match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
WARN  [lis3mdl] no device on bus 2
WARN  [hmc5883] no device on bus 2 (type: 1)
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] probe failed! 255
WARN  [mpu9250] no device on bus 4
MPU6000 on SPI bus 1 at 4 (1000 KHz)
WARN  [mpu6000] no device on bus 5
WARN  [mpu9250] probe failed! 104
WARN  [mpu9250] no device on bus 3
WARN  [mpu9250] probe failed! 255
WARN  [mpu9250] no device on bus 4
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
Airspeed on I2C bus 1 at 0x28 (bus: 100 KHz, max: 100 KHz)
INFO  [load_mon] stack check enabled
INFO  [px4io] default PWM output device
ERROR [px4io] already loaded
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
WARN  [uavcan] Node ID 1, bitrate 1000000
WARN  [uavcan] sensor bridge 'gnss' init ok
WARN  [uavcan] sensor bridge 'mag' init ok
WARN  [uavcan] sensor bridge 'baro' init ok
INFO  [ver] match: PX4FMU_V2
px4flow [229:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [logger] logger started (mode=all)
Unknown MAV_TYPE
WARN  [uavcan] GNSS ORB fd 10
INFO  [init] Mixer: /etc/mixers/quad_x_can.main.mix on /dev/uavcan/esc
WARN  [uavcan] subscribe to actuator_controls_0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
***WARN  [navigator] task start failed
***ERROR [navigator] start failed

NuttShell (NSH)
nsh> free
             total       used       free    largest
Mem:        234416     231824       2592       2240
nsh> param show SYS*
***nsh: param: command not found
nsh> ver all
HW arch: PX4FMU_V2
FW git-hash: 0d22e97c09e493559b8c9645bea21e4cd7c53077
FW version: 1.6.0 0 (17170432)
OS: NuttX
OS version: Release 1.8.0 (17301759)
Build datetime: Apr 25 2017 08:42:04
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
MFGUID: 303533333036510400410026
MCU: STM32F42x, rev. 3
UID: 410026:30365104:30353333
```

pavel-kirienko, 2017-04-26:
Consider invoking `uavcan shrink` and see if it changes anything.

LorenzMeier, 2017-04-30:
@pavel-kirienko Could you provide a patch that does this inside the UAVCAN app at the right time?

LorenzMeier, 2017-04-30:
Without UAVCAN running:

```
Processes: 23 total, 2 running, 21 sleeping
CPU usage: 64.97% tasks, 1.58% sched, 33.46% idle
DMA Memory: 5120 total, 1536 used 1536 peak
Uptime: 1028.490s total, 384.803s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  384802 33.458   644/  748   0 (  0)  READY 
   1 hpwork                      42252  4.201   992/ 1780 192 (192)  w:sig 
   2 lpwork                       2615  0.225   656/ 1780  50 ( 50)  w:sig 
   3 init                        32248  0.000  2056/ 2484 100 (100)  w:sem 
 204 commander_low_prio             36  0.000   592/ 2996  50 ( 50)  w:sem 
  95 gps                          2499  0.225   808/ 1476 220 (220)  w:sem 
 103 dataman                       180  0.000   704/ 1180  90 ( 90)  w:sem 
 143 sensors                     51393  5.026  1328/ 1980 250 (250)  w:sem 
 145 commander                   21990  2.025  2896/ 3548 140 (140)  w:sig 
 152 px4io                       32454  3.000   952/ 1380 240 (240)  w:sem 
 165 mavlink_if0                 20245  1.800  1736/ 2436 100 (100)  w:sig 
 166 mavlink_rcv_if0               444  0.075  1208/ 2100 175 (175)  w:sem 
 178 mavlink_if1                 14244  1.575  1712/ 2388 100 (100)  w:sig 
 179 mavlink_rcv_if1              1958  0.150  1320/ 2100 175 (175)  w:sem 
 342 top                            76  2.850  1288/ 1684 100 (100)  RUN   
 238 log_writer_file                 0  0.000   344/ 1060  60 ( 60)  w:sem 
 218 mavlink_if2                137928 13.878  1712/ 2388 100 (100)  w:sig 
 219 mavlink_rcv_if2               159  0.000  1328/ 2100 175 (175)  w:sem 
 235 logger                       5955  0.525  1312/ 3532 250 (250)  w:sem 
 286 ekf2                       188664 18.904  5112/ 5780 250 (250)  w:sem 
 296 mc_att_control              47223  4.651  1256/ 1676 250 (250)  w:sem 
 298 mc_pos_control              56469  5.701  1240/ 1876 250 (250)  w:sem 
 309 navigator                    2137  0.150  1272/ 1572 105 (105)  w:sem 
nsh> free
             total       used       free    largest
Mem:        236864     206000      30864      25312
```

LorenzMeier, 2017-05-01:
Resolved by: https://github.com/PX4/Firmware/pull/7166

closed 2017-05-01 by LorenzMeier

pavel-kirienko, 2017-05-02:
Sorry that it took me forever to respond, but my proposed solution wouldn't have helped anyway. According to the boot log posted above, the memory issues begin immediately after the UAVCAN node is started. These issues could not be induced by dynamic memory allocation, because in the very short time interval between initialization of the UAVCAN node and the moment where issues begin the node would not have the opportunity to use much of dynamic memory anyway.

Normally the UAVCAN node never frees the heap memory it acquires; instead, it takes 64 byte sized chunks from the heap as needed and stores them indefinitely for future use. The `shrink` command forces the node to `free()` all memory chunks that are not in use at the moment. The downside of this action is that it introduces a short-term performance penalty and increases memory fragmentation, so normally it should not be invoked at all, unless heap reserves become really scarce.