PX4/PX4-Autopilot #7128: Takeoff w/o GPS works in Gazebo, but not in real life

TobiMiller, 2017-04-25:
Hey,

why can i takeoff w/o GPS in SITL but it doesnt work in real enviroment? I want to start autonomously and feed external vision pose (marker tracking) after takeoff into the px4. In my gazebo simulation everything works fine.

dagar, 2017-04-26:
What doesn't work? Are failsafes being triggered?

jgoppert, 2017-04-28:
A log would be helpful.

Seanmatthews, 2017-06-12:
I'm going to try to consolidate all the other similar issues into this newest one, and go from there.

Relevant:
- [This comment](https://github.com/PX4/Firmware/issues/7055#issuecomment-295638659) infers that the commander interprets a lack of GPS as a lack of position.
- #7266 is related in that GPS timeout seems to inhibit certain OFFBOARD functionality.
- #7279 describes POS_HOLD rejection when GPS is unavailable.
- (Honorable mention) #7089 doesn't contain enough info, but it's a GPS fault when using LPE.

If the linked comment still holds true, then @dagar is correct in that failsafes, due to lack of GPS, are being triggered. Apart from rewriting the commander, does anyone on this thread have an idea of which parts of PX4 we would target to either abstract the GPS position throughout the commander, or a more temporary fix for LPE until the commander rewrite happens? I'm happy to make the changes, but some guidance would be helpful for sure.

mhkabir, 2017-06-12:
@ChristophTobler already has the stuff necessary to solve this in a branch : https://github.com/PX4/Firmware/tree/global_to_local. It would need testing with LPE because it's only been tested with EKF2.

Aside : I'm reworking some parts of EKF2 this week to allow an external estimator (VIO or whatever) to be fused in the loop.

Seanmatthews, 2017-06-12:
That's great news. @ChristophTobler, is there anything I should know in order to test out your branch with LPE? Is it ready to be tested with LPE, or are there additional issues to deal with?

ChristophTobler, 2017-06-12:
@Seanmatthews It is still work in progress and it has only been tested in SITL. So if you do want to test it, make sure you have manual mapped on your remote in case somethings happens

Seanmatthews, 2017-06-12:
@ChristophTobler What's the best way to map a Spektrum controller to switch between manual and offboard (or mission?) ? Looking [here](https://pixhawk.org/dev/manual_input) it appears that channel 5 switches between manual, assisted, and mission by default (though my channel 5 is mapped to only a two-way switch). What's the common safety mechanism/control settings that PX4 devs use for a deadman's switch?

ChristophTobler, 2017-06-14:
What ever works best for you, I guess. Just make sure you have manual and a kill switch enabled.

Seanmatthews, 2017-06-15:
@ChristophTobler Does the branch allow external position estimates to LPE? Through mavros, I send VIO position estimates to topic `/mavros/vision_pose/pose_cov` but I see no callback on `/mavros/local_position/pose` (position coming from FCU). Essentially, what I get is a diagonal takeoff where the quadcopter keeps moving forward, but it was commanded to go to (0, 0, 1). Here's [video](https://streamable.com/u2tyr). The same code in simulation has the vehicle going to altitude with a bit of drift, so I think maybe PX4 is not getting the external position estimation. The VIO gives an accurate position. Any thoughts? What additional information can I obtain for you?

TSC21, 2017-07-25:
@TobiMiller @Seanmatthews does those problems still persist?

Seanmatthews, 2017-07-25:
@TSC21 Not for me. My problem was due to an unrelated issue and may be disregarded.

closed 2017-07-27 by TSC21

TobiMiller, 2017-07-28:
Everything works fine for me. Sorry for not editing this.