PX4/PX4-Autopilot #7131: VTOL - motors from turning off in FW

OJdoesIt, 2017-04-26:
My aircraft has a Y3 motor config. While in Manual mode (for testing), when I switch from MC -> FW, all the motors turn off. If I move the throttle, the front motors start to spin again, basically reacts to the throttle. When I switch back to MC, all the motors automatically turn back on.
How can I prevent the motors from turning off when in FW mode?

dagar, 2017-04-26:
Which airframe did you choose?

OJdoesIt, 2017-04-26:
I am using my custom airframe, which used mixer adapted from Eflite Convergence. But instead of elevons, I use ailerons. I also made the adjustment within the mixer.

dagar, 2017-04-26:
Rereading your comment again, that's exactly what should be happening. Manual mode FW is like RC, with all the controls passed through. If you used a mode like altitude control then the throttle would be automatically set to maintain airspeed and altitude.

OJdoesIt, 2017-04-27:
I just tested by switching to Altitude mode and and the motors stop spinning in FW. The throttle stick has no effect neither. Is there a parameter I'm missing?

dagar, 2017-04-27:
No throttle at all in manual MC? The motors are still spinning at idle though?

In FW altitude control mode the throttle will stay at idle (a param) if the vehicle thinks it's landed. The throttle stick is now sending airspeed commands.

OJdoesIt, 2017-04-27:
I just tested the manual MC and it works. I just didn't work when I had no GPS signal. I'll test the FW issue and get back to you.

dagar, 2017-04-30:
It sounds like this is more of an issue with the autopilot not providing good feedback? If you can post thorough details we can improve it.

closed 2017-04-30 by dagar