PX4/PX4-Autopilot #7132: Submodule update Nuttx problem

kyuhyong, 2017-04-26:
Hello,
I recently cloned master branch into my pc (both in Ubuntu and mac) as shown below

> dronedev@dronedev-MXC6300:~/workspace$ git clone https://github.com/PX4/Firmware.git ./px4Firmware
> Cloning into './px4Firmware'...
> remote: Counting objects: 208111, done.
> remote: Compressing objects: 100% (42/42), done.
> remote: Total 208111 (delta 13), reused 0 (delta 0), pack-reused 208068
> Receiving objects: 100% (208111/208111), 84.14 MiB | 29.00 KiB/s, done.
> Resolving deltas: 100% (155736/155736), done.
> Checking connectivity... done.
> 

And tried make px4fmu-v2_default and failed due to git diffs. 

> |diff --git NuttX/nuttx/include/signal.h NuttX/nuttx/include/signal.h
> |index bd8ee9f..7ee242a 100644
> |--- NuttX/nuttx/include/signal.h
> |+++ NuttX/nuttx/include/signal.h
> --------------------------
> File to patch: [  4%] Built target nuttx_copy_px4fmu-v2
> Scanning dependencies of target px4fmu-v2-nuttx_patch_00001-REJECTED-add-math.h.patch
> [  4%] px4fmu-v2: nuttx-patches/00001-REJECTED-add-math.h.patch applied
> [  4%] Built target px4fmu-v2-nuttx_patch_00001-REJECTED-add-math.h.patch
> Scanning dependencies of target px4fmu-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch
> [  5%] px4fmu-v2: nuttx-patches/00002-REJECTED-fix-shadow-wanings.patch applied
> The text leading up to this was:
> --------------------------
> |diff --git NuttX/nuttx/include/signal.h NuttX/nuttx/include/signal.h
> |index bd8ee9f..7ee242a 100644
> |--- NuttX/nuttx/include/signal.h
> |+++ NuttX/nuttx/include/signal.h
> --------------------------
> File to patch: [  5%] Built target xml_gen

The reason was the Nuttx folder in the root is empty. 
I did submodule init and update but I when cloning Nuttx, I got followings.

> dronedev@dronedev-MXC6300:~/workspace/px4Firmware$ git submodule init
> Submodule 'NuttX' (https://github.com/PX4/PX4NuttX.git) registered for path 'NuttX'
> Submodule 'Tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'Tools/gencpp'
> Submodule 'Tools/genmsg' (https://github.com/ros/genmsg.git) registered for path 'Tools/genmsg'
> Submodule 'Tools/jMAVSim' (https://github.com/PX4/jMAVSim.git) registered for path 'Tools/jMAVSim'
> Submodule 'Tools/sitl_gazebo' (https://github.com/PX4/sitl_gazebo.git) registered for path 'Tools/sitl_gazebo'
> Submodule 'cmake/cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon) registered for path 'cmake/cmake_hexagon'
> Submodule 'mavlink/include/mavlink/v1.0' (https://github.com/mavlink/c_library_v1.git) registered for path 'mavlink/include/mavlink/v1.0'
> Submodule 'mavlink/include/mavlink/v2.0' (https://github.com/mavlink/c_library_v2.git) registered for path 'mavlink/include/mavlink/v2.0'
> Submodule 'src/drivers/gps/devices' (https://github.com/PX4/GpsDrivers.git) registered for path 'src/drivers/gps/devices'
> Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) registered for path 'src/lib/DriverFramework'
> Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) registered for path 'src/lib/ecl'
> Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) registered for path 'src/lib/matrix'
> Submodule 'src/modules/uavcan/libuavcan' (https://github.com/UAVCAN/libuavcan.git) registered for path 'src/modules/uavcan/libuavcan'
> Submodule 'unittests/googletest' (https://github.com/google/googletest.git) registered for path 'unittests/googletest'
>  dronedev@dronedev-MXC6300:~/workspace/px4Firmware$ git submodule update
> Cloning into 'NuttX'...
> remote: Counting objects: 290, done.
> remote: Compressing objects: 100% (20/20), done.
> remote: Total 290 (delta 10), reused 0 (delta 0), pack-reused 270
> Receiving objects: 100% (290/290), 39.00 KiB | 0 bytes/s, done.
> Resolving deltas: 100% (119/119), done.
> Checking connectivity... done.
> Submodule path 'NuttX': checked out '44ad7e224c1ef17911ab8b4101fd624ad9ee4177'

It looks like the git version of NuttX is different from the master branch of PX4 Firmware. 
Manually downloading NuttX cannot solve this issue. 
What have to be done to "make" work again? 
Thanks,

Kyu

kyuhyong, 2017-04-26:
Sorry, it's my fault.
I should have git update as per the instruction as here. 
git submodule update --init --recursive

closed 2017-04-26 by kyuhyong

dagar, 2017-04-28:
Not a problem. We have helper targets to work around git submodule pain.

    make submodulesclean

    make distclean