PX4/PX4-Autopilot #7133: HIL coordinates jumps

borune-k12, 2017-04-26:
When in HIL mode with XPlane vehicle's coorinates jumps.

borune-k12, 2017-04-26:
[image]

dagar, 2017-04-26:
Is HIL successfully connected to xplane at this point?

borune-k12, 2017-04-26:
what do you mean successfully? In "HIL config" widget XPlane frequency is updating..and when plane is on ground there are no jumps..it appears when plane is in air

Antiheavy, 2017-04-26:
@borune-k12 what is the update rate from Xplane?  I've noticed similar behavior sometimes when the update rate is low (e.g. < 30 Hz) and the EKF get really unhappy.
[image]

borune-k12, 2017-04-26:
@Antiheavy it varies from 20 to 30 Hz. I dont think that it's the cause of problem since with the stable firmware frequency is the same but jumps don't happen.

dagar, 2017-04-30:
@borune-k12 which version of PX4 is this?

borune-k12, 2017-05-02:
@dagar i dont know what the number but i got it from qgc from master branch

pgorczak, 2017-11-28:
I'm encountering the same behavior sometimes. A workaround is making sure that the simulated airplane is parked and not moving much, then rebooting (or re-connecting the USB) of the px4. When hitting connect in the HIL window and waiting for a couple seconds, the state in QGC should be stable and reflecting the state in X-Plane.

closed 2018-02-28 by PX4BuildBot