PX4/PX4-Autopilot #7136: Missing camera triggers in distance based triggering

Antiheavy, 2017-04-26:
It seems PX4 is occasionally missing camera triggers in distance base triggering.  We've seen this happen both in real flight testing and also in HIL simulation.   FMUv4 based hardware, MAVlink compatible camera on Telem2.  Any ideas why we might be seeing this or what we could look into to further diagnose?

Actual flight:
[image]

HIL sim flight:
[image]

Another HIL sim flight:
[image]

mhkabir, 2017-04-26:
Is this in mission mode ?

mhkabir, 2017-04-26:
It would be good to debug this over a call or IM. Can we Skype?

Antiheavy, 2017-04-26:
yep

mhkabir, 2017-04-26:
I'm 'kabiriris' on Skype.

dagar, 2017-04-26:
Are these plots from autopilot logs or camera?

LorenzMeier, 2017-04-26:
Adding to Daniel: Can you share an autopilot log from one of these plots?

Antiheavy, 2017-04-26:
The plots above are from our camera log.  Attached are the SDcard logs from the two HIL missions above.

[HIL-logs-maybe-missing-camera-triggers.zip](https://github.com/PX4/Firmware/files/959743/HIL-logs-maybe-missing-camera-triggers.zip)

mhkabir, 2017-04-26:
> The plots above are from our camera log.

Hmm, that seems suspect. We could potentially be losing trigger packets over the link, and thus they come up as 'missing' triggers. I will add logging for the trigger uORB message on a branch (we don't log those ATM) and send you a link/binaries for testing in HIL. You would need to use the new logger.

mhkabir, 2017-04-26:
Ok we aren't really missing trigger events on the PX4 side. This is the trigger sequence number plotted against time. I realised that @Antiheavy's sdlog2 logs also have the trigger events logged. I added it at some point and then forgot :P  

[image]

LorenzMeier, 2017-04-27:
@mhkabir So the conclusion is that we're not dropping trigger events but some drop on the way to the ground?

LorenzMeier, 2017-04-27:
@Antiheavy Are you manually enabling the "CAMERA_TRIGGER" trigger message somewhere? If you do, how does the line look like? What is your telemetry link?

mhkabir, 2017-04-27:
@LorenzMeier after the discussion with @Antiheavy, it seems that the companion link to the camera gets throttled because it's running only at 57600 baud.

Antiheavy, 2017-04-27:
Our camera talks to the FMUv4 over telem2 UART that we have set to Companion 57600.  We look for DO_DIGICAM_CONTROL messages and execute triggers based on those.  Next week we hope to set up a test to figure out exactly where the messages are getting lost.  Will report back when we know more.

mhkabir, 2017-04-28:
You must have some customization on your firmware side to send `DO_DIGICAM_CONTROL` messages?

You should use the `CAMERA_TRIGGER` message instead, which should be sent without throttling and by default : https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2043-L2044

dagar, 2017-04-28:
I don't think this is a PX4 issue. Let's followup on skype for mavlink debugging.

closed 2017-04-28 by dagar

Antiheavy, 2017-06-15:
For what it is worth, we found the telem2 serial buffer was overflowing with the default telemetry.  Dialing back the data rate and number of messages seems to have fixed this - at least in HIL sim.  We will fly it soon.

mhkabir, 2017-06-15:
@Antiheavy We also added a display for camera trigger points in PX4 Flight Review (logs.px4.io) now, so you can actually see where the triggering system commands a shot if you upload the log.