PX4/PX4-Autopilot #7137: navio2 limited to 4 PWM output

nicolaerosia, 2017-04-26:
Hello,

I'm building a hexacopter+ using Navio2 on RPi3 and I noticed that the pwm driver is limited to 4 outputs because of https://github.com/PX4/Firmware/blob/master/src/drivers/navio_sysfs_pwm_out/navio_sysfs_pwm_out.cpp#L60 and https://github.com/PX4/Firmware/blob/master/src/drivers/navio_sysfs_pwm_out/navio_sysfs_pwm_out.cpp#L352-L362 .
I changed 4 to 6 to work in my case but I would like to fix it for everyone.
MixerGroup->mix returns the number of outputs, but that is called in the loop and the initialization is done before that.

Any thoughts?

bkueng, 2017-04-27:
I think we can increase it to the maximum (14?). Do you want to open a PR?

Looking at it, I see there's already a bug: `_outputs.noutputs` is passed to `pwm_limit_calc`, which can be >4, as `_outputs.NUM_ACTUATOR_OUTPUTS = 16` is passed to `mix()`.

nicolaerosia, 2017-04-27:
The only problem I have with 14 is that it would prohibit the usage of pwm channels as AUX. I need to look how other drivers providing AUX support are implemented.

bkueng, 2017-04-28:
How do you intend to use the AUX channels? If you use an aux mixer file, you can just append the file content to the main mixer file. Other drivers like the fmu support the `MIXERIOCLOADBUF` ioctl's to load mixers via `mixer load ...` or via `mixer append ...` commands.
We can also add a startup parameter to the driver, specifying the max number of channels it should use.

nicolaerosia, 2017-05-10:
I don't plan to use AUX channels, I don't know how they are used, I was thinking maybe other user need it. Anyway, this issue will be fixed by your PR, https://github.com/PX4/Firmware/pull/7200

bkueng, 2017-05-10:
Good to know. I'm closing this then.

closed 2017-05-10 by bkueng