PX4/PX4-Autopilot #7140: Mapping RC input to AUX not working on MindPX

mzahana, 2017-04-26:
Hi,

I used to map RC channels to some AUX pins on px4fmu-v2, to control e.g. a gimbal. That worked fine. Now I am trying to do the same on MindPX v2.8. However, it seems that it's not yet implemented. This, I guess should be a basic feature. So, can some one help explaining how to do this on MindPX, or give some directions, and maybe I can try to implement it.
Thanks.

dagar, 2017-04-26:
@rolandash ping

iZhangHui, 2017-04-29:
Pixhawk is the all-in-one unit, combining FMU and IO into a single package, while MindPX(FMUV4 family) only have a FMU as showed below: 
[image]
We have been trying to implement both of MAIN/AUX PWM functions on MindPX. Honestly, this is a big challenge to us.
So far, we have a basic and rough codes for this target. Once pass the basic testing, we will issue a PR for review.

mzahana, 2017-04-30:
@iZhangHui Thanks for the explanation. Waiting for the PR.

mzahana, 2017-05-08:
@izhanghui is there any updates on this?

tubeme, 2017-05-09:
3 months ago I tried forwarding some rc to aux with regular Pixhawk fmuv2 and it did not work as well... The issue I posted https://github.com/PX4/Firmware/issues/6827. There might be some more basic problem with the mixer...

mzahana, 2017-05-09:
I had no problems with regular pixhawk recently. I am not able to do this
on MindPX.

iZhangHui, 2017-05-10:
@mzahana Could you do a test for PR #7208 ?

iZhangHui, 2017-05-10:
Built firmware for MindPX
[mindpx-v2_default.px4.zip](https://github.com/PX4/Firmware/files/988898/mindpx-v2_default.px4.zip)

mzahana, 2017-05-10:
@iZhangHui Thanks! that PR solved the problem. I can set the the pass through from QGC, and control gimbal from my RC.
My question now is can that be used with the vmount (or gimbal app) to stabilize a gimbal similar to a Pixhawk?

iZhangHui, 2017-05-11:
Yes, the behavior is similar to Pixhawk.
https://dev.px4.io/en/advanced/gimbal_control.html

closed 2018-02-21 by PX4BuildBot