PX4/PX4-Autopilot #7142: PX4IO-V2 incredibly low on stack space again

NaterGator, 2017-04-26:
I am finding I can't load an 8 channel SimpleMixer configuration, I'm out of RAM after loading the first 5 channels. 

@davids5 I'm wondering if this has to do with the NuttX change and the switch to CONFIG_ARCH_INTERRUPTSTACK. That's the largest global symbol by far.

davids5, 2017-04-26:
Hi @NaterGator  - 

Yes It is suspect. The 2 things I can share are: 

1)  If there is more then one task, I think having the one stack is better that having N space on all stacks. Where N is max ISR save + call depth in ISR.  

If there is just the one task we should just use CONFIG_ARCH_INTERRUPTSTACK=0 (At the time, I based the [change on](https://github.com/PX4/Firmware/blob/master/nuttx-configs/px4io-v2/nsh/defconfig#L620) CONFIG_MAX_TASKS=4 not a review of the PX4IO subsystem.

2) I ran into an alignment issue I have not been able to convince the linker to get over!

A recent io_timer_changed added 8 bytes to the bss. This moved the stack allocation up by 252 bytes and for some reason I have not been able to find or change it. The +8 realigned the stack allocation at the next page boundary (256) and wasted 252 bytes.  (I have tried assuming .align n was `1<< n and [tried this]( https://github.com/PX4/Firmware/commit/79c5fc31f4189d7cbccc0e01cd6fa671e600aaee)

So as a Q&D fix I did [this](https://github.com/PX4/Firmware/commit/6ef2ae29993c02f926c8719d29d84202b1fe3142)!
 To get past "Police Lights"

I have not had time to circle back. So  your expert eyes looking at this would be very much appreciated!

You may be able to reduce the the other stacks or that buffer a bit again, just check  the symbols  first to see if there is any more wasted space from the [lack of]( https://github.com/PX4/Firmware/commit/79c5fc31f4189d7cbccc0e01cd6fa671e600aaee)

LorenzMeier, 2017-05-02:
Merged https://github.com/PX4/Firmware/pull/7167 and consequently closing this.

LorenzMeier, 2017-05-02:
Actually keeping it open as a "new" issue to track memory management for IO on next release.

davids5, 2017-05-02:
I have one more "high hanging fruit" I want to pick. 
I will be forced to do it for our next TBD update to upstream of NuttX any way.  
It will take some work, but I described it [here](https://groups.yahoo.com/neo/groups/nuttx/conversations/messages/14649)  

I think I can get back another ~228 bytes on current master's Nuttx and twice that on upstream NuttX

LorenzMeier, 2017-05-26:
@NaterGator Any news on this?

NaterGator, 2017-06-07:
Lorenz, things are looking good now! Closing.

closed 2017-06-07 by NaterGator