PX4/PX4-Autopilot #7147: RTL will transition to MC regardless of the NAV_FORCE_VT parameter

hamishwillee, 2017-04-27:
RTL will transition to MC regardless of the NAV_FORCE_VT parameter: https://github.com/PX4/Firmware/blob/master/src/modules/navigator/rtl.cpp#L326-L328

Not sure if this is intended or a bug - but either the implementation or [docs](https://github.com/PX4/Devguide/issues/132) need to be updated.

sanderux, 2017-04-27:
The NAV_FORCE_VT parameter was not intended for RTL behavior, it was intended for accidental LAND instructions or manually switching to land mode.

The reason RTL forces this is people might want FW landings enabled but in an RTL event that doesnt always work. The behavior was indeed intended, perhaps it could be debated.

hamishwillee, 2017-04-27:
OK. Worth adding this clarification in the [parameter doc](https://dev.px4.io/en/advanced/parameter_reference.html#NAV_FORCE_VT) ? E.g. 
FROM 
NAV_FORCE_VT - Force VTOL mode takeoff and land.
TO
NAV_FORCE_VT - Force VTOL for takeoff and land (but not RTL).

sanderux, 2017-04-28:
@dagar do you think RTL should honor NAV_FORCE_VT? or do we keep it hardcoded on land as MC?

dagar, 2017-04-28:
I personally don't know any VTOL users that would even want/need NAV_FORCE_VT configurable.
I think we should leave it as is, unless someone specifically asks for this behaviour.

sanderux, 2017-04-28:
I agree, @hamishwillee please update the docs

closed 2017-04-28 by sanderux

hamishwillee, 2017-04-30:
@dagar Can you update the parameter doc (if you think necessary) to

> NAV_FORCE_VT - Force VTOL for takeoff and land (but not RTL).

I'll fix up our other ones.