PX4/PX4-Autopilot #7150: ulog is not working properly on MindPX v2.8

mzahana, 2017-04-28:
Hi,

I am using MindPX v2.8 that is bought very recently. I am using it on Hexarotor, and it flies ok. However, I am not able to record ulog file when I choose the new logger option in QGC. Is it not supported on MindPX. I am using latest  PX4 master as of now. Thanks.

mzahana, 2017-04-28:
@dagar were you able to reproduce the issue?

mzahana, 2017-04-28:
Another note is, I used QGC to download the logs, which are not working anyway, and didn't use QGC to erase them. However, when I pulled out the SD card and plugged it to my PC, I see no logs in the SD card! Does QGC erase logs automatically after downloading?

mzahana, 2017-04-28:
Here is a pic of the SD card. The ```log``` folder is empty !
[image]

TSC21, 2017-04-28:
I will try to reproduce this ASAP

LorenzMeier, 2017-04-30:
You need to use a recent version from master to use the new logger, because it was off by default on MindPx.

closed 2017-04-30 by LorenzMeier

mzahana, 2017-04-30:
How recent is recent? I tried this couple of days ago and did not work.
Anyway will try with  latest master again.

mzahana, 2017-04-30:
* yesterday, I changed SD card, and used master branch that is couple of days old from the time of this comment, and got *ulog working.
* now, i just updated to latest master, and got *ulog too.

I am guessing it is an SD card problem, although it's a new SD that cam with MindPX!