PX4/PX4-Autopilot #7156: Multiple Erle-Copters

AliSiddiqui4, 2017-04-28:
Hi, 

I managed to spawn 3 erle-copters in Gazebo by modifying the erle-copter spawn launch file. But as there is only 1 MAVlink so once I try hovering the quads all three of them go hay wire and ust fly away randomly. 

How to make separate MAV for each robots or any other alternatives?

sanderux, 2017-04-28:
change each config file to run mavlink on a different port

AliSiddiqui4, 2017-04-28:
@sanderux  Hi, 
I am really new at this. I know about the .launch file and the xacro and urdf file. Can you please explain where is this config file, and how do I modify it?

sanderux, 2017-04-28:
I'm not sure how erlecopter packaged it. Perhaps you better ask them.
Within px4 that would be here: https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris#L65

But perhaps the discussion forum or erlecopter is a better location as this is not an issue related to px4

closed 2017-04-28 by sanderux