PX4/PX4-Autopilot #7158: Lidar-Lite V3 - PWM interface not working

tomatac, 2017-04-30:
I am trying to achieve position control using a PX4FLOW and a Lidar-Lite V3 and it looks like the PWM interface is not working. I used the latest master branch as of today APR/29.
I followed these instructions: https://pixhawk.org/peripherals/rangefinder
I checked the wiring and I updated the wiring diagram for Lidar-Lite V3 (see the picture below).
To test/debug the Lidar-lite connectivity on nsh I tried:

nsh> pwm_input start
ERROR [pwm_input] driver already started

nsh> pwm_input test
Showing samples for 5 seconds

nsh> ll40ls start pwm
ERROR [ll40ls] driver already started

nsh> ll40ls test pwm
ERROR [ll40ls] immediate read failed
ERROR [ll40ls] Task exited with errno=11

nsh> ll40ls info pwm
state @ 20010120
ll40ls_pwm_read: 124635 events, 8477306us elapsed, 68us avg, min 29us max 2522us 154.253us rms
ll40ls_pwm_read_errors: 124635 events
ll40ls_pwm_zero_resets: 0 events
WARN  [ll40ls] poll interval:  50 ticks
WARN  [ll40ls] distance: 0.000m

[image]

mhkabir, 2017-04-30:
Do you have a BEC connected on the servo rails to power the Lidar?

tomatac, 2017-04-30:
No. I do not have any other servo. Do I need to add a zener diode if I only power the Lidar Sensor on the servo rail? Is there a schematic on the recommended wiring for the servo rail? Thanks!

mhkabir, 2017-04-30:
You shouldn't need anything else. Just plug in the BEC on the servo rail. 
Also, you could just use i2c; The v3 Lidars work fine with it.

closed 2017-04-30 by mhkabir