PX4/PX4-Autopilot #7159: Build firmware error

Ashucool41, 2017-04-30:
I am tryiing to build by following instructions from https://dev.px4.io/en/setup/building_px4.html but got these errors (on Ubuntu 16.04 LTS):

make px4fmu-v2_default
-- PX4 VERSION: v1.6.0rc1-959-g0a80ee6
-- CONFIG: nuttx-px4fmu-v2-default
-- CMAKE_MODULE_PATH: /home/ashu/src/Firmware/cmake
-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- The ASM compiler identification is GNU
-- Found assembler: /home/ashu/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
CMake Error at CMakeLists.txt:319 (STRING):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.

-- C compiler: 
CMake Error at CMakeLists.txt:327 (STRING):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.

-- C++ compiler: 
-- Adding ROMFS on px4fmu-v2
-- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
-- Configuring incomplete, errors occurred!
See also "/home/ashu/src/Firmware/build_px4fmu-v2_default/CMakeFiles/CMakeOutput.log".
/bin/sh: 1: cd: can't cd to /home/ashu/src/Firmware/build_px4fmu-v2_default
Makefile:161: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

LorenzMeier, 2017-04-30:
Please update your toolchain installation. Since you're using GCC 4.9 you can't be using up-to-date build instructions - those would have installed GCC 5.4. You might be missing a whole list of other things as well.

closed 2017-04-30 by LorenzMeier

Ashucool41, 2017-04-30:
I am using GCC 5.4 only and also double checked every other packages given on the website. Is there any other package I need to install ?

Ashucool41, 2017-04-30:
And both the above mentioned errors correspond to this particular line in the cmakelists.txt file:
" STRING(REGEX MATCH "[^\n]*" c_compiler_version_short ${c_compiler_version})"

mhkabir, 2017-04-30:
> I am using GCC 5.4 only 

No you're not. See your own build output.

dagar, 2017-04-30:
@Ashucool41 which version of cmake is this? You likely need to update the dev environment, but I'd like the cmake config to fail more gracefully with useful messages.

Can you print `${CMAKE_C_COMPILER}` and `${c_compiler_version}` here? -
 https://github.com/PX4/Firmware/blob/master/CMakeLists.txt#L318

Ashucool41, 2017-04-30:
I meant I'm getting the same error even after updating

make px4fmu-v2_default
-- PX4 VERSION: v1.6.0rc1-959-g0a80ee6
-- CONFIG: nuttx-px4fmu-v2-default
-- CMAKE_MODULE_PATH: /home/ashu/src/Firmware/cmake
-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- The ASM compiler identification is GNU
-- Found assembler: /root/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
CMake Error at CMakeLists.txt:319 (STRING):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.

-- C compiler: 
CMake Error at CMakeLists.txt:327 (STRING):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.

-- C++ compiler: 
-- Adding ROMFS on px4fmu-v2
-- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
-- Configuring incomplete, errors occurred!
See also "/home/ashu/src/Firmware/build_px4fmu-v2_default/CMakeFiles/CMakeOutput.log".
/bin/sh: 1: cd: can't cd to /home/ashu/src/Firmware/build_px4fmu-v2_default
Makefile:161: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

dagar, 2017-04-30:
Can you print those variables I mentioned above? Which version of cmake?

andresR8, 2017-07-21:
@Ashucool41  maybe you forgot to run this:
`sudo dpkg --add-architecture i386`
`sudo apt-get update`
`sudo apt-get install libc6:i386 libgcc1:i386 libstdc++5:i386 libstdc++6:i386`
`sudo apt-get install gcc-4.6-base:i386`