PX4/PX4-Autopilot #7168: HC_SR04 driver needs cleanup

LorenzMeier, 2017-05-02:
@davids5 The HC_SR04 driver needs cleanup. I'm not asking you to do it, just track it. We should point the Airmind developers to that.

LorenzMeier, 2017-05-02:
Main questions:

  * Is the driver functional?
  * Why does it have so much code commented out?
  * The chinese comments should all be translated to english
  * How has this been tested?

davids5, 2017-05-02:
1. [Fix](https://github.com/PX4/Firmware/blob/master/src/drivers/hc_sr04/hc_sr04.cpp#L198-L204) Driver should use the timer capture interface IOCTL See FMU::test.

2. [Fix](https://github.com/PX4/Firmware/blob/master/src/drivers/hc_sr04/hc_sr04.cpp#L541-L545) usleep will not work here. 

The OS granularity is 1 ms. Priority and quantisation errors will make this a very long pulse.  
Options:
a) prefered: Use the PWM to trigger it.
b) OK: Do something clever.  Only set the pin here, Then clear it on the measure.
c) BAD: disable ints set, use up_udelay(10) clear, enable ints.

3. [Fix](https://github.com/PX4/Firmware/blob/master/src/drivers/hc_sr04/hc_sr04.cpp#L545) Use the input capture.

Use input capture because it a more immune to interrupt latency and jitter.  Using px4_arch_gpiosetevent is not good. because:

```
Pulse to measure _________-------_______________
Other ISRs disable----------___________-------------
Time measured with 
GPIO   ___________________--------------_____________________

Time measured with Capture 
call back_________________-_____________-_____________________
Time measured with Delta T is correct but delayed but Other ISRs disable time
Capture__________________________-------_______________
```

rolandash, 2017-05-04:
This driver was functional - on earlier MindPX hardware though, before V2.3 i think. It is in preliminary status. It aims to support up to maximum 6 sonars, directly through MindPX FMU's I/O pin, save the needs for an external co-processor for range finders.

After that as there are major relocations of hardware I/O resources i think it not working on MindPX after V2.6 yet. 

As we are trying to add AUX support to MindPX now - so this driver needs to be adjusted to work with the new AUX architecture. Thanks @davids5 for the advice, PWM is also what we prefered. MindPX V2.8 hardware has led out all 16 PWM channels of total 4 timers, so this can be interesting. 

And probably this can be extended to other type of range finders like LiDAR lite, especially can benefit those only have PWM interfaces.
@LorenzMeier @iZhangHui

LorenzMeier, 2017-10-04:
Fixed meantime.

closed 2017-10-04 by LorenzMeier