PX4/PX4-Autopilot #7179: Pixhawk Pro: inspect communication with IO

bkueng, 2017-05-03:
When testing the Pixhawk Pro I ran into a state where communication with the IO failed. I don't remember the exact steps, I did some `reboot`s on the console before it happened. This is the output:
```
WARN  [px4io] fmu sent: "0
S: 3 5 10000 10000 0 -10000 10000
M: 1
O: 10000 10000 0 -00;"
px4io mixer send error -1
```
Subsequent reboots showed CRC errors from px4io and mixer loading failures.
I could however not reproduce it after power cycling.

LorenzMeier, 2017-05-03:
Could very well be this one:
https://github.com/PX4/Firmware/pull/7172

LorenzMeier, 2017-05-04:
I'm closing this since this seems to be an exact duplicate of an existing issue. Please re-open if you can reproduce it after flashing master from *right now*.

closed 2017-05-04 by LorenzMeier