PX4/PX4-Autopilot #7183: Bit-banding for atomic flag access on STM32

LorenzMeier, 2017-05-04:
By using bit-banding we could avoid having to lock the OS for flag accesses to avoid read-modify-write issues: http://www.micromouseonline.com/2010/07/14/bit-banding-in-the-stm32/

LorenzMeier, 2017-05-04:
Starting point would be to replace these: https://github.com/PX4/Firmware/pull/7181/commits/93bcde197ddf4b9eea31cc3b82dd8a37c3bae5dd#diff-8c9290622c738995a6aa1c57f5d7706dR206

davids5, 2017-05-04:
I woulsd avoid BB at it is not supported on all STM32's 
How about built in atomic  https://gcc.gnu.org/onlinedocs/gcc-4.6.1/gcc/Atomic-Builtins.html know to work on ARM

bkueng, 2017-05-05:
BB is indeed not portable. I already looked into atomic operations (for RCU) and planned to introduce a general API for PX4 based on GCC's builtins. They are supported by Clang as well.

closed 2017-10-05 by LorenzMeier