PX4/PX4-Autopilot #7184: problem in Building PX4

hamididari, 2017-05-04:
hi guys 
I try to building px4 but i got this err

`[  0%] Built target ver_gen
[  0%] Built target git_nuttx
[  0%] Built target git_gencpp
[  0%] Built target git_genmsg
[  0%] Built target git_matrix
[  0%] Built target git_uavcan
[  0%] Built target git_mavlink
[  0%] Built target mixer_gen
[  1%] Built target xml_gen
[  2%] Built target git_ecl
[  3%] Built target nuttx_copy_px4io-v2
[  3%] Built target px4io-v2-nuttx_patch_00001-REJECTED-add-math.h.patch
[  3%] Built target nuttx_copy_px4fmu-v2
[  3%] Built target px4io-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch
[  3%] Built target px4fmu-v2-nuttx_patch_00001-REJECTED-add-math.h.patch
[  4%] Built target px4io-v2-nuttx_patch_00003-REJECTED-add-var-expansion-in-nsh-parse.patch
[  4%] Built target px4fmu-v2-nuttx_patch_00002-REJECTED-fix-shadow-wanings.patch
[  4%] Built target px4io-v2-nuttx_patch_00004-REJECTED-avoid-export-copy-with-export-insitu.patch
[  4%] Built target px4fmu-v2-nuttx_patch_00003-REJECTED-add-var-expansion-in-nsh-parse.patch
[  4%] Built target px4io-v2-nuttx_patch_00005-REJECTED-support-c++11.patch
[  4%] Built target px4fmu-v2-nuttx_patch_00004-REJECTED-avoid-export-copy-with-export-insitu.patch
[  4%] Built target px4io-v2-nuttx_patch_00006-REJECTED-cstdint-fix.patch
[  5%] Built target px4fmu-v2-nuttx_patch_00005-REJECTED-support-c++11.patch
[  5%] Built target px4io-v2-nuttx_patch_00007-REJECTED-silence-jobserver-warnings.patch
[  5%] Built target px4fmu-v2-nuttx_patch_00006-REJECTED-cstdint-fix.patch
[  5%] Built target px4io-v2-nuttx_patch_00008-REJECTED-static-assert-fix.patch
[  5%] Built target px4fmu-v2-nuttx_patch_00007-REJECTED-silence-jobserver-warnings.patch
[  6%] Built target px4io-v2-nuttx_patch_00009-REJECTED-ctype-fix-shadow-wanings.patch
[  6%] Built target px4fmu-v2-nuttx_patch_00008-REJECTED-static-assert-fix.patch
[  6%] Built target px4io-v2-nuttx_patch_00010-BACKPORT-stm32-flash-F4-dcache-corruption-fix-no-HSI-on.patch
[  6%] Built target px4fmu-v2-nuttx_patch_00009-REJECTED-ctype-fix-shadow-wanings.patch
[  6%] Built target px4io-v2-nuttx_patch_00011-BACKPORT-priority-restoration-fix.patch
[  6%] Built target px4fmu-v2-nuttx_patch_00010-BACKPORT-stm32-flash-F4-dcache-corruption-fix-no-HSI-on.patch
[  6%] Built target px4io-v2-nuttx_patch_00012-BACKPORT-stack-coloration-overreach-fix.patch
[  7%] Built target px4fmu-v2-nuttx_patch_00011-BACKPORT-priority-restoration-fix.patch
[  7%] Built target px4io-v2-nuttx_patch_00013-BACKPORT-stm32-serial-dma-hotfix.patch
[  7%] Built target px4fmu-v2-nuttx_patch_00012-BACKPORT-stack-coloration-overreach-fix.patch
[  8%] Built target px4io-v2-nuttx_patch_00014-BACKPORT-i2c-hotfix.patch
[  8%] Built target px4fmu-v2-nuttx_patch_00013-BACKPORT-stm32-serial-dma-hotfix.patch
[  8%] Built target px4io-v2-nuttx_patch_00015-BACKPORT-stm32f7-DTCM.patch
[  8%] Built target px4fmu-v2-nuttx_patch_00014-BACKPORT-i2c-hotfix.patch
[  8%] Built target px4io-v2-nuttx_patch_00016-BACKPORT-fix-CRTSCTS-defines.patch
[  8%] Built target px4fmu-v2-nuttx_patch_00015-BACKPORT-stm32f7-DTCM.patch
[  8%] Built target px4io-v2-nuttx_patch_00017-BACKPORT-cdcacm.patch
[  8%] Built target px4fmu-v2-nuttx_patch_00016-BACKPORT-fix-CRTSCTS-defines.patch
[  8%] Built target px4io-v2-nuttx_patch_00018-BACKPORT-stm32-serial-break.patch
[  9%] Built target px4fmu-v2-nuttx_patch_00017-BACKPORT-cdcacm.patch
[  9%] Built target px4io-v2-nuttx_patch_00019-BACKPORT-stm32-rcc-keep-HSI-on.patch
[  9%] Built target px4fmu-v2-nuttx_patch_00018-BACKPORT-stm32-serial-break.patch
[ 10%] Built target px4io-v2-nuttx_patch_00020-BACKPORT-ramtron-CONFIG-prefix.patch
[ 10%] Built target px4fmu-v2-nuttx_patch_00019-BACKPORT-stm32-rcc-keep-HSI-on.patch
[ 10%] Built target px4io-v2-nuttx_patch_00021-BACKPORT-stm32f3x-add-BKP.patch
[ 10%] Built target px4fmu-v2-nuttx_patch_00020-BACKPORT-ramtron-CONFIG-prefix.patch
[ 10%] Built target px4io-v2-nuttx_patch_00022-BACKPORT-stm32-bkp-reference-fix.patch
[ 10%] Built target px4fmu-v2-nuttx_patch_00021-BACKPORT-stm32f3x-add-BKP.patch
[ 10%] Built target px4io-v2-nuttx_patch_00023-BACKPORT-stm32f7-bkp-reference-fix.patch
[ 11%] Built target px4fmu-v2-nuttx_patch_00022-BACKPORT-stm32-bkp-reference-fix.patch
[ 11%] Built target px4io-v2-nuttx_patch_00024-BACKPORT-stm32f7-serial-dma-hotfix.patch
[ 11%] Built target px4fmu-v2-nuttx_patch_00023-BACKPORT-stm32f7-bkp-reference-fix.patch
[ 11%] Built target px4io-v2-nuttx_patch_00025-BACKPORT-add-set-ex-to-nsh.patch
[ 11%] Built target px4fmu-v2-nuttx_patch_00024-BACKPORT-stm32f7-serial-dma-hotfix.patch
[ 12%] Built target px4io-v2-nuttx_patch_90000-PENDING-wip-inflight-to-upstream.patch
[ 12%] Built target px4fmu-v2-nuttx_patch_00025-BACKPORT-add-set-ex-to-nsh.patch
[ 12%] Built target nuttx_patch_px4io-v2
[ 12%] Built target px4fmu-v2-nuttx_patch_90000-PENDING-wip-inflight-to-upstream.patch
[ 12%] Built target nuttx_configure_px4io-v2
[ 12%] Built target nuttx_patch_px4fmu-v2
[ 12%] Built target nuttx_export_px4io-v2
[ 12%] Built target nuttx_configure_px4fmu-v2
[ 12%] Built target io_prebuild_targets
[ 12%] Built target nuttx_export_px4fmu-v2
[ 12%] Built target prebuild_targets
[ 13%] Generating uORB topic headers
[ 12%] Generating uORB topic multi headers for nuttx
[ 13%] Generating uORB topic sources
python import error:  No module named em

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

python import error:  No module named em

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

python import error:  No module named em

Required python packages not installed.

On a Debian/Ubuntu system please run:

  sudo apt-get install python-empy
  sudo pip install catkin_pkg

On MacOS please run:
  sudo pip install empy catkin_pkg

On Windows please run:
  easy_install empy catkin_pkg

CMakeFiles/msg_gen.dir/build.make:153: recipe for target 'topics_sources/uORBTopics.cpp' failed
make[3]: *** [topics_sources/uORBTopics.cpp] Error 1
make[3]: *** Waiting for unfinished jobs....
CMakeFiles/msg_gen.dir/build.make:528: recipe for target 'src/platforms/nuttx/px4_messages/px4_actuator_armed.h' failed
make[3]: *** [src/platforms/nuttx/px4_messages/px4_actuator_armed.h] Error 1
CMakeFiles/msg_gen.dir/build.make:900: recipe for target 'src/modules/uORB/topics/actuator_armed.h' failed
make[3]: *** [src/modules/uORB/topics/actuator_armed.h] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/msg_gen.dir/all' failed
make[2]: *** [CMakeFiles/msg_gen.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:161: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2
`
i try to fix the problem with this command :
  sudo apt-get install python-empy
  sudo pip install catkin_pkg

`hamid@hamid:~/Firmware$   sudo apt-get install python-empy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-empy is already the newest version (3.3.2-1build1).
The following packages were automatically installed and are no longer required:
  libmbim-glib4 libmbim-proxy libqmi-glib1 libqmi-proxy ubuntu-core-launcher
  usb-modeswitch usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 12 not upgraded.
hamid@hamid:~/Firmware$   sudo pip install catkin_pkg
The directory '/home/hamid/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/hamid/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied (use --upgrade to upgrade): catkin_pkg in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): argparse in /usr/lib/python2.7 (from catkin_pkg)
Requirement already satisfied (use --upgrade to upgrade): docutils in /usr/local/lib/python2.7/dist-packages (from catkin_pkg)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/lib/python2.7/dist-packages (from catkin_pkg)
You are using pip version 8.1.1, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
`

dagar, 2017-05-08:
Which version of Ubuntu is this? Did you follow these instructions? https://dev.px4.io/en/setup/dev_env_linux.html

Make sure you install all the packages, reboot to make sure $PATH is updated, then start with a clean PX4 repo. `make distclean`

dagar, 2017-05-08:
Please ask on http://discuss.px4.io/ or https://gitter.im/PX4/Firmware if you continue to have issues.

If you find the documentation is out of date or missing something it would be awesome if you could help fix it here. https://github.com/PX4/devguide

closed 2017-05-08 by dagar

djfly, 2018-06-17:
I encountered the same problem, I found that this is because my system has two python, delete one to solve, in order to facilitate the people who encountered the same problem, hereby reply

suraj2596, 2018-12-19:
Try this : `pip3 install toml`