PX4/PX4-Autopilot #7185: Losing GPS does not trigger fail safe

ChristophTobler, 2017-05-04:
On latest master, failsafe when losing GPS does not work. A few test cases:
* When I unclick `use GPS` for the EKF2_AID_MASK (in air), the only thing appearing is `WARN  [navigator] global position timeout` after a few seconds but nothing happens. The quad just starts to drift.
* When I stop sending GPS (in air), nothing happens. The quad just starts to drift.
* When the home position was set but has no GPS anymore (on ground), I can still takeoff. Then the quad drifts

I guess the EKF reports local position as still valid?

ChristophTobler, 2017-05-04:
@priseborough FYI

ChristophTobler, 2017-05-04:
for the first case `_ekf.local_position_is_valid()` reports invalid, hence the navigator warning. For the other two cases local position is still valid.

ChristophTobler, 2017-05-04:
Ok, so I don't think it's the estimator but commander/navigator because I was disabling GPS here https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L572. With `gpssim stop` the local and global position are invalid but landing does not work.

ChristophTobler, 2017-05-04:
The error seems to be in here https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3779

ChristophTobler, 2017-05-04:
@priseborough It will never enter https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L3786.  `required_accuracy * 2.5f` is too big because `local_position->eph` will only increase to about `6`. After https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L438-L448 it is `0`.

It works if I remove the `2.5` multiplier but I´m not sure if this is very robust.

priseborough, 2017-05-04:
OK, I see the problem, will submit a PR soon.

closed 2017-05-15 by ChristophTobler