PX4/PX4-Autopilot #7194: MC 1.6RC YAW Fault when switch to Loiter during Mission

tops4u, 2017-05-05:
Today I wanted to test https://github.com/mavlink/qgroundcontrol/issues/4902 which basically works now. But when switching to Loiter during Mission the Vehicle turned around by about 160degress. Continuing the mission, the YAW Angle stayed like this. I assume this to be a bug in PX4.

Check the Log : http://logs.px4.io/plot_app?log=c0cf8e2d-b2c5-4594-9a09-764ee2210ded

tops4u, 2017-05-10:
Just to add one more thing. While looking at the above log it seems to have been truncated. After Mission Flight I was flying PosCTL to the Starting Point and landed there. This is completely Missing in the Log.

LorenzMeier, 2017-05-18:
@tops4u Does this branch fix it for you? https://github.com/PX4/Firmware/pull/7260

tops4u, 2017-05-18:
I will test this today evening. I assume this does not fix the YAW behaviour?

LorenzMeier, 2017-05-20:
@tops4u Did you get around to test?

tops4u, 2017-05-20:
@LorenzMeier terribly sorry but not yet. I'll have a live Demo tomorrow and I'm away today so I hope to be able to Install and Test maybe on Sunday or Monday evening so you should get Feedback from me before the next TelCo.

LorenzMeier, 2017-05-20:
Not release gating. Will look into this next week.

tops4u, 2017-05-22:
No news for the Serial yet (not updated yet), but a MIssion Flight today and the YAW Problem did not happen?!?

http://logs.px4.io/plot_app?log=a280ae70-50b9-4276-b79f-dfed3799f14e

tops4u, 2017-05-23:
Today I was flying with latest Master (from yesterday). Frozen Stream and YAW Turn occurred both again. 

Log: http://logs.px4.io/plot_app?log=67cfc7e3-3f79-4c72-8d6d-f892dc5cb717

Also I had the impression, that Log Download was slower than before - but that may also be related to my Laptop.

closed 2018-02-21 by PX4BuildBot