PX4/PX4-Autopilot #7196: feature request: starting tone for compass calibration

Antiheavy, 2017-05-06:
There is an audible double-tone from the autopilot when completing a spin around an axis during compass calibration - this is nice feedback.

Would it be possible to add a single tone to indicate an axis is ready to start spinning?  Right now we have to watch the screen in QGC for feedback.  Because you want to stand far away from metal and electronics when doing a field compass calibration this can be difficult.

Thanks in advance for your consideration.

dagar, 2017-05-08:
In general we should try to make it possible to perform all calibrations solo (not having to watch the screen). Things like feedback for starting and finishing a side, but also a negative tone if you accidentally repeat. 
What about QGC audio?

Antiheavy, 2017-05-10:
QGC audio would be useful too.  Although if outdoors and standing 10 meters away from the ground station while spinning the vehicle it might be nice to have the aircraft itself beep too.  I think having audible tones both places would be great.

Would PX4 send a text string to QGC or would QGC itself do the heavy lifting here?

Antiheavy, 2017-07-09:
Thanks @dagar. 
https://github.com/PX4/Firmware/pull/7558

closed 2017-07-09 by Antiheavy