PX4/PX4-Autopilot #7198: Generic Quad X - Boot Up Problem

cagdas001, 2017-05-07:
The device does not boot up after selected the Generic Quad X as the airframe in the latest stable release. There is no problem on other frame types, the only problem is with Generic Quad X.

dagar, 2017-05-08:
Are you able to connect a console and see where it's getting stuck?

cagdas001, 2017-05-09:
Since I calibrated everything and mounted on the frame, I don't want to reset the board again. When you set Generic Quad X, the device does not open at all and therefore you can't connect it to QGC, you can't install the firmware again. I think it may be related to the bootloader. Because I was able to install ArduCopter firmware with the MissionPlanner.

But I can say there is no problem in beta and dev releases. I had tried them also.

dagar, 2017-05-09:
Oh, in that case this will be resolved shortly. Use dev (master) for now, it's going to become the next release version very soon.

closed 2017-05-09 by dagar