PX4/PX4-Autopilot #7202: RTL never completed on RC loss, resulting in crash

tstellanova, 2017-05-08:
Log here: http://logs.uaventure.com/view/Zqps3y3ZWeMmfXezj8GfCa

- Pixhawk 2.4 configured to RTL and land on RC loss
- Vehicle was switched into position hold
- RC transmitter turned off (intentionally)
- Pixhawk began RTL sequence but paused over Home at safe RTL altitude 
- After waiting for vehicle to descend, vehicle continued to slowly ascend 
- To avoid flyaway, vehicle was emergency disarmed from QGC, causing flight termination.

dagar, 2017-05-08:
@tstellanova can you tell us the closest PX4 commit?

@Stifael @MaEtUgR could you take a look at this MC flyaway in AUTO LAND?

[image]

tstellanova, 2017-05-09:
Closest commit would be 4a36a806ded2d4fc79c52dc3cc7f12938567e4a9
Thanks for taking a look...

LorenzMeier, 2017-05-10:
@priseborough Can you have a look?

MaEtUgR, 2017-05-10:
- Why does the main_state of the commander (not nav_state) change to LAND in RTL?
I mean this is not logically wrong but I've never seen this before.

- In the end we have a pretty small velocity setpoint of 0.25 m/s downwards that is reported to be tracked not very good but surely not in the wrong direction. An the drone goes up at the same time. This is what we've seen before here's a related pr: https://github.com/PX4/Firmware/pull/7144

- What I can also see in the log that there are regularly slowdowns in log point update rates. Do we know if that's only the log or are there even sensor EKF slowdowns that could lead to an integration error and therefore worse bias of velocity? (this last one is just a guess)

tops4u, 2017-05-11:
@MaEtUgR I can add to your last Point, that I have reported a similar behaviour (Logging slowdown) and IMHO it does at least impact Mavlink since I get Messages on the OSD indicating, that the MavLink Stream is Frozen -> See #6260

priseborough, 2017-05-11:
The root cause was poor quality IMU hardware. There is a large change in Z accel bias through flight that the EKF was unable to keep up with. Here is the low pass filtered Z accel:

[image]

I am unable to find IMU temperature anywhere in the log, but it is likely due to thermal drift. Here are the vertical velocity and position innovations:

[image]

The controller was unable to handle the 0.6 m/s vertical velocity error that resulted. The most recent version of the ecl EKF publishes an alternate vertical velocity measure that is the first derivative of the height output. The control loops should be using that if the height rate set point saturates. See prototype here:

https://github.com/priseborough/Firmware/tree/position-derivative-wip

Longer term I think we should move to tracking a reference point at all times using accel feed-forward in addition to velocity and position error feedback wiht a leash to handle saturation. The current approach of switching between velocity and position control depending on a large number of parameters is proving problematic.

The effect of such a large bias rate of change on the estimator could be reduced by increasing the value EKF2_ACC_B_NOISE.

tops4u, 2017-05-11:
@priseborough could you elaborate if this happens mostly due to (Temperature) Drift or due to high vibrations? As I use a very light AUAV-X2 controller, I have problems limiting the amount of vibration on it, but I did perform a Temperature Compensation run. 

This is the Accel Z Log:
[image]

LorenzMeier, 2017-05-11:
You should be concerned - but not because the code isn't stable, but because your system has a vibration level that makes it impossible to work for an inertial navigation system.

@priseborough we need to significantly improve the vibration reporting in v1.6 I think.

LorenzMeier, 2017-05-11:
I'm closing this meanwhile because this is a mechanical setup issue.

priseborough, 2017-05-11:
In this instance the source problem is poor thermal stability of the IMU Z accel. There is no evidence of any thermal calibration having been performed - the TC parameters are at default values.

LorenzMeier, 2017-05-11:
Yep - I mistook the 8 for a 0 on my phone, the noise is much less excessive than I thought.

tops4u, 2017-05-11:
@LorenzMeier & @priseborough thank you for your Feedback. So I take this as: Vibrations of my Vehicles IMU are far from perfect but are within acceptable Range?

tstellanova, 2017-05-11:
Thanks for the analysis @priseborough .  Indeed we performed a standard accel calibration on the pixhawk but did not perform the thermal calibration. Is there any monitoring in place to detect excessive noise in the IMU? I'd like to understand how we can detect this situation in the future, even after performing thermal calibration.

priseborough, 2017-12-10:
https://review.px4.io/plot_app?log=f619bcb1-7427-4619-86d6-bc53e5e2024e

priseborough, 2017-12-10:
The vertical position derivative data required to ensure vertical velocity estimation errors due to bad accel bias values do not prevent landing is available in the local_position message. See the plot of vz and z_deriv from the above crash log

[image]

A simple change in the position controller would prevent this type of this crash.

priseborough, 2017-12-10:
This log shows large vertical innovations and delta velocity bias state estimates present at start of flight. This would have been caught by updated pre-flight checks. 

**vertical innovations**
[image]

**delta velocity bias estimates**
[image]

Without preflight logging data it is impossible to tell how this error condition was initiated.

priseborough, 2017-12-10:
The parameter value for the pre-flight max bias check is 0.005 which is greater than the max range used by the estimator. I will push a change to this parameter to that the check will trigger if the takeoff bias estimates exceed 50% of the max range allowed by the estimator. This will ensure that the estimator has enough margin to accomodate further changes in bias due to inflight temperature variation.

priseborough, 2017-12-11:
The following PR's have been raised to address the inflight vulnerability to vertical velocity estimation offsets: https://github.com/PX4/Firmware/pull/8445 , and the ineffectiveness of the current pre-flight delta velocity bias check: https://github.com/PX4/Firmware/pull/8446

LorenzMeier, 2017-12-31:
Closing as fixed.

closed 2017-12-31 by LorenzMeier