PX4/PX4-Autopilot #7203: Fix use of the copy constructor and initialization

LorenzMeier, 2017-05-08:
IIRC we are using in a number of places the copy constructor to assign an empty object. That's not super efficient and should be changed.

We also initialize fields in many places where they do not need this, resulting in excessive overhead. This is in particular true in the MAVLink app.

dagar, 2017-05-08:
If you point out the copy constructor abuse we could temporarily (or permanently) disable it to catch all the instances.

For orb_copy we could try to make more use of the Subscription class to discourage this pattern. For mavlink we might also want to look at sharing the objects.

The problem is automatically catching the initialization cases that actually matter. The static analysis tools apply guidelines regardless, and exhaustively checking the codebase using the dynamic tools isn't really feasible. Structuring the code differently to avoid this situation might be the best option.

dagar, 2017-05-15:
Note for later, look into c++ move semantics (move constructor + move assignment operator).

closed 2018-02-21 by PX4BuildBot