PX4/PX4-Autopilot #7211: Navio2 battery

nicolaerosia, 2017-05-10:
Hello,

Navio2 does not have support in PX4 for Battery.
Example code for reading the ADC is [here](https://docs.emlid.com/navio2/dev/adc/)
Looking through the code, I see two systems support battery:
1. px4io
2. df_bebop_bus_wrapper

It seems to me that I need to create an app, e.g. navio2_batt, which will do the following:
a) subscribe to actuator_controls_0, since updateBatteryStatus needs throttle_normalized
b) read the voltage & current
c) update an instance of Battery using updateBatteryStatus
d) publish battery_status

Is this correct?

bkueng, 2017-05-11:
Yes that sounds correct. There is a second approach: create a driver that provides an ioctl interface, such as [adcsim](https://github.com/PX4/Firmware/blob/master/src/platforms/posix/drivers/adcsim/adcsim.cpp) or [aerofc_adc](https://github.com/PX4/Firmware/blob/master/src/drivers/aerofc_adc/aerofc_adc.cpp), which is then used by the sensors app, and publishes battery_status (remove the [ifdef](https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L557)). If the battery parameters are used ([here](https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L487)), I'd say use this solution, otherwise either way should be fine.

nicolaerosia, 2017-05-14:
I've created a PR for this here https://github.com/PX4/Firmware/pull/7236

nicolaerosia, 2017-05-15:
The PR was merged

closed 2017-05-15 by nicolaerosia