PX4/PX4-Autopilot #7212: Discontinuity in raw mag readings

CarlOlsson, 2017-05-10:
I set the calibration coefficients to default for the external mag and then did a mag calibration while logging. When inspecting the log there is a discontinuity in the readings. 
[image]
This was on modified firmware, however I remember seeing the same thing in [this log](http://logs.px4.io/plot_app?log=9ea27c0a-5c42-4f30-920e-4f5095dbafe0) reported [here](http://discuss.px4.io/t/crash-px4-fw-1-6-0-rc/3220)
[image]

Anyone know why this happens and if it is intended behavior?

julianoes, 2017-05-12:
@CarlOlsson is the discontinuity when you do the calibration? This would make sense because the offsets are set to 0 for the duration of the calibration and then the new offsets are applied.

CarlOlsson, 2017-05-12:
For us it happens during calibration every time if you before set all mag0 cal parameters to default. However, I am just pulling in some changes from PX4 master and will see if that fixes it. For the log I posted above from the 1.6 RC I guess it could be a switch from external mag to an internal? Unfortunately the sensor_selection topic isn't logged so hard to tell

julianoes, 2017-05-12:
But that's expected, I'm confused.

CarlOlsson, 2017-05-15:
I just pulled latest master, reset all mag0 calibration parameters to default, rebooted and did a calibration. Here is the plot from the log file
[image]
As can be seen in the plot there is a spike in the beginning of the calibration routine at around 60s. @julianoes is this expected?

julianoes, 2017-05-15:
Oh, you're saying it spikes at the very beginning of the calibration. Ok, no that doesn't seem expected.
@bkueng have you seen this?

closed 2018-02-21 by PX4BuildBot