PX4/PX4-Autopilot #7213: MAV_CMD_SET_CAMERA_MODE in mission fails with unknown command

DonLakeFlyer, 2017-05-10:


dagar, 2017-05-10:
I don't see it implemented or part of an open PR. @mhkabir is this planned?

mhkabir, 2017-05-10:
Not from my side. We need to crystallise the API first anyway. Nothing uses this command yet.

dogmaphobic, 2017-05-10:
It's there alright. Just not updated within the firmware:

http://mavlink.org/messages/common#MAV_CMD_SET_CAMERA_MODE

DonLakeFlyer, 2017-05-11:
I talked to Lorenz about this yesterday. QGC currently works as follows:
If you add a survey to a mission. QGC will automatically add a SET_CAMERA_MODE to photo to the beginning of the mission.

My question to Lorenz was whether it was OK to do this in all cases for PX4. He said yes, and that he could clean up the unknown command coming back from firmware when QGC does this.

dagar, 2017-05-11:
Oh okay, do we need to accept it or simply not respond unsupported?

DonLakeFlyer, 2017-05-11:
It should be supported by PX4 I think, but I don't know how all that works. @LorenzMeier would know better what he was thinking.

LorenzMeier, 2017-05-11:
We just don't NACK it for now.

mhkabir, 2017-05-11:
So the `SET_CAMERA_MODE` on QGC master breaks PX4 currently (Mission rejected). We need to roll back QGC or bring in transitional support, because it would break all older PX4 versions even if we were to add the command to the ignore/no-NACK list in v1.6.

@LorenzMeier relevant to out discussion today about non-lockstep GCS/FW combos.

DonLakeFlyer, 2017-05-11:
Yeah, I'll get right on that.

DonLakeFlyer, 2017-05-11:
> So the SET_CAMERA_MODE on QGC master breaks PX4 currently (Mission rejected).

That would be the exact reason I talked to Lorenz about this in the first place. Users of old firmware can turn off the photo mode switch in QGC. In general there is a problem with mavlink in that QGC has no way to ask the firmware what commands are supported or unsupported. It can only keep a hardcoded list directly in the code. Which it does, but that list is global to firmware type and not version specific.

mhkabir, 2017-05-11:
> Yeah, I'll get right on that.

@DonLakeFlyer Thanks! Meanwhile, I've fixed camera triggering with transitional support so that we don't affect older setups using Survey.

mhkabir, 2017-05-11:
I have added transitional 1.6 release support for `SET_CAMERA_MODE` (ignored but accepted for now) in my camera trigger PR.

DonLakeFlyer, 2017-05-11:
The other strangeness that comes up with this sort of thing is. Yes, the firmware may support the command. But the camera may not. Or the vehicle itself might not even have a camera on it. That all gets tricky as to how to make any sense out of that from the QGC side of things which has little information to go on.

LorenzMeier, 2017-05-12:
Closing consequently as that got merged.

closed 2017-05-12 by LorenzMeier