PX4/PX4-Autopilot #7220: command denied during calibration: 205/204

cobya, 2017-05-12:
After cloning the most recent build and building the firmware, I am getting the error "command denied during calibration: 205/204" when attempting to calibrate any sensor onboard the Pixhawk board. This happens 6 times during the calibration of the accelerometer, or constantly (approx every 1/20th of a second) with the gyroscope / compass. This has happened on two separate pixhawks and a variety of airframe configurations, both stock and custom during my testing. I have uploaded an image of the critical messages shown in QGroundControl 3.1.3.

[image]

dagar, 2017-07-29:
Could you try again with the latest PX4/QGC?

closed 2018-02-21 by PX4BuildBot

tomzzz989, 2022-07-03:
I also encountered this problem. I am 1.11.0. How do you solve it