PX4/PX4-Autopilot #7221: Actuator Outputs Bounded After Saturation Causes Vehicle Decent

lamping7, 2017-05-12:
Log shows the actuator output being bounded between a min and max (1400 and 1820, not PWM params) after saturation. Max is not enough to hover, causing a forced land. Stabilized flight mode. My understanding is that the mixer should scale the outputs so that the mc is stable (roll and pitch) first, then take care of yaw, with the goal of the actuators being scaled between 1 and 0, where the saturated motor would be 1, not some fraction less than that.

**Log:** http://review.px4.io/plot_app?log=a041097e-dac2-4f50-9062-ff6b4287e3d6

sanderux, 2017-05-12:
Your thrust and manual throttle input never exceeded 65%
is that correct?

lamping7, 2017-05-12:
@sanderux Correct for RC input. MPC_THR_HOVER = 0.7 due to heavy platform weight. Saturation occurred as RC throttle was increased while maneuvering. RC throttle continued to increase and actuator output then became bounded to the lesser value, from 1400 and 1820. Data shows an increase in thrust, but vehicle descends. Why would the actuator output be bounded in this manner? Looking at https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp I can't see why this would occur.

Here is a part of the graph I'm concerned about:
[Actuator Output Graph](https://i.imgur.com/coOCdpm.png)

lamping7, 2017-05-13:
Happened again during testing today. 

http://review.px4.io/plot_app?log=14373214-034e-44e7-bfaa-5074182def3d

Less platform weight - MPC_THR_HOVER reduced to 0.5. This situation was a bit odd but had a similar outcome. Vehicle flew far out. Went out of range of the RC. After 0.5 seconds of loss, loiter was triggered as the failsafe. Pilot wasn't aware when link was regained. The mc began descent (which I assume is the result of the RC input). Looking at the plots, you can see that about 5m off the ground the pilot increased throttle to return to hover. PWM out became saturated and then bounded again.

sanderux, 2017-05-13:
From what i can tell the saturation is related to yaw. It will not use the whole range for yaw as it would have nothing left for thrust. @dagar do you know how this is mixed on an octo?

lamping7, 2017-05-13:
After looking at a different flight log, I'm not sure it is related to saturation. Here is another log from yesterday where the went from AUTO.TAKEOFF to STABILIZED. After takeoff, the pilot took over and throttled up, causing saturation. You can see one motor hit 1950. The same bounding see in the previous 2 logs did not occur. Not sure if this helps in figuring out what is causing this.

http://review.px4.io/plot_app?log=23974268-5adb-4295-9515-9f63da0ba4b5

sanderux, 2017-05-14:
To the best of my knowledge there is a limit being applied to how much actuation is being used for yaw, roll and pitch. As far as i can tell the saturation you are seeing is related the the large error in yaw. You could try to lower your yaw rate P and I

lamping7, 2017-05-14:
@sanderux Yaw angle P and FF were already lowered. We could try lowering yaw rate gains. In both cases the RC was not commanding yaw till later. The command was coming from the FC. And the log shows the yaw angular rate diverging from the setpoint by >25 deg/s.

My concern is: Why would the output be bounded in a way that is **persistent** and which limits **all** actuator control including thrust.  By reading the mixer algorithm, it looks as though the intent is to mix in yaw last so that yaw does not effect thrust, roll or pitch commands in the case of saturation.

sanderux, 2017-05-14:
From hoe i read the log there was no thrust demand and in stab mode it will not generate that demand for you

sanderux, 2017-05-14:
From your logs i do not see it limiting thrust. I do get the feeling your
drone is underpowered and the CG is not correct. Have you tried flying
lighter and have you performed tuning on your frame?

Regards,
Sander Smeets

lamping7, 2017-05-15:
@sanderux We did lower the platform weight. If you look between my comments 2 and 3, the 3rd is much less, which is why the MPC_THR_HOVER was brought down to 0.5. Roll and pitch PID gains were tuned not buy guess during flight, but by fixing the vehicle to a stand that allowed for movement and by reviewing log data over many iterations. 

I can confirm that the FC does not sit exactly at GC (GC is lower, as the batteries sit in the tray below, the FC is in plane with the motors), but I still do not understand why the actuator outputs are being bounded in a way that persists though the rest of the flight. In both logs you can see that RC commands an increase in throttle, but shortly after PWM actuator outputs become bounded. What causes this bounding? If it is not implied, thanks for looking at this.

diegoeck, 2017-05-19:
From what I understand, you have a large yaw error, and the controller is putting 4 rotors at a large value (~1800), and 4 rotors in a small value (~1400). The point is that the PWM x thrust is not linear, and even that the mean value (1600) in all motors could lift the copter, half in 1800 and half in 1400 can not lift.

 I believe that the loss of lift is due line 343 (thrust -= thrust_reduction;). There is a thrust reduction to allow more yaw (but maybe there is something wrong). Why is this thrust_reduction necessary? Line 336 puts the output at maximum. Why should we reduce the output at line 343?

https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp

diegoeck, 2017-05-19:
Also, line 343 computes yaw several times (for each rotor) but i rewrites the value, and at end "yaw" has the value computed for the last rotor. It seems strange to me..  @sanderux  can you check line 343 and comment? (in fact line 308-340).

LorenzMeier, 2017-05-19:
My experience with Mixing is consistently that people underestimate it and by the time they have seen all side effects they leave as is.

Would anyone on this thread be up to write a proper unit test? Then we can have a data driven discussion and risk massive changes and see the effects.

dagar, 2017-05-19:
I can help by putting the initial structure in place for mixing unit tests, but it's not yet clear to me how to make them meaningful for understanding these cases.

diegoeck, 2017-05-19:
I can help also, but I'm not sure how to proceed...

LorenzMeier, 2017-05-19:
@dagar Can you help just putting up the structure with the right files and inputs and outputs and a shell command? It should be as lightweight as possible. Then @diegoeck should be able to generate inputs and test the outputs.

mhkabir, 2017-05-30:
We encountered it on two airframes (VTOL and MC). This should be fixed by https://github.com/PX4/Firmware/pull/7310, which is now in master.

Please test @lamping7. Please reopen if still an issue.

closed 2017-05-30 by mhkabir

lamping7, 2017-05-31:
Thanks guys. I will not be able to test for a few weeks.